pyscada/PyScada #1: Modbus communication

DavidGib, 2016-04-14:
Hello, I want to create a HMI for Siemens PLC thanks to PyScada with windows. I wonder how the modbus connection is set and how the variables are changed. I change the values on HMI with the command "$(".type-numeric.var-key").html(Value);" but it doesn't send on modbus. Can you help me ?
Thank you !

trombastic, 2016-04-29:
Hello David,

the documentation of the backend is currently in progress and i hope that all your questions regarding the setup of modbus devices and the HMI will be answered soon. Have you some further information for me what is working and what's not in your setup, do you use modbus/ip or rtu, what version of pyscada have you installed and what version of windows do you use.

DavidGib, 2016-05-03:
Thank you for your answer. 
I can change the values of my variables on the html panel and they are also changed in the control panels. But I don't know how to get from the modbus communication and change them. I created modbus variables related to the variables I use in the HMI and I did a server module in the PLC. I use modbus IP with Pyscada 0.6 on windows 7.

SHIVAPRASAD96, 2017-03-30:
After installing all dependencies , When head towards Command-line and then ,

ISSUE 1 :
When i run the Command :  **service gunicorn_pyscada start** 
I am getting the following Error:
**gunicorn_pyscada: unrecognized service**

ISSUE 2:
when i run : ** systemctl start "pyscada_*  **
i get **Failed to get D-Bus connection: Failed to connect to socket /run/systemd/private: No such file or directory
**
Kindly Respond for the above Queries , Eagerly waiting for the answers because company project has been halted for the same.

singhaisid0, 2017-06-21:
Hello David,
I wonder how modbus is connected with pyscada.Can you help me in this.

closed 2018-04-17 by trombastic