pyscada/PyScada #2: Charts

semargal, 2016-11-19:
Hello!
How can I build charts with this library? There is nothing about how to do that http://pyscada.readthedocs.io/en/dev-0.7.x/backend.html#add-a-new-chart
Is it possible or maybe is planned?

trombastic, 2017-01-17:
Dear semargal,

yes it is possible to add add Charts to the HMI. I will update the Docs as soon as possible, may be the short instructions for Version 0.7.x will help you to get it work in the meantime.

In the Backend HMI Section (http://IP/admin/hmi/):
1. Charts, add a new Chart
2. Page, add a Page
3. Widget, add a Widget, select under Page the page you added in 2. and under Chart the Chart from 1., a widget controls the position of every element on a Page.
4. View, add a View and select the page from 2.
5. GroupDisplayPermissions, add a new GroupDisplayPermission, (if nessesary add a new Group and add your User to that Group, select all items you created in 1. to 4.
6. open http://IP/, you should see the new View, if the DAQ is running and there is Data already in the DB, you should see the last 2 Houers of Data and the curennt Data.

```
+-View------------------------------------+
|                                         |
| +-Page--------------------------------+ |
| |                                     | |
| | +-Widget--------+ +-Widget--------+ | |
| | |               | |               | | |
| | | Row 1, Col 1  | | Row 1, Col 2  | | |
| | | +-Chart-----+ | | +-Chart-----+ | | |
| | | |           | | | |           | | | |
| | | +-----------+ | | +---------- + | | |
| | +---------------+ +---------------+ | |
| +-------------------------------------+ |
+-----------------------------------------+
```

semargal, 2017-02-05:
Thank you for your answer!

closed 2017-02-05 by semargal