pyscada/PyScada #6: Updated documentation

cwraig, 2017-12-15:
I am unsure if this is the correct place to post this but I have written a series of blog posts for documenting my installation of pyscada on raspberrypi. I would be happy to have this incorporated into the documentation associated with the project. Let me know if you are interested and I could have a look at moving the relevant components over.

Thanks for all the work that has gone into the project, i hope to have a look into the modbus RTU side in the next week or so. 

https://cwraig.id.au/2017/09/17/pyscada-on-raspberry-pi-for-temperature-monitoring-with-ds18b20-on-1-wire-part-1-software-installation/

https://cwraig.id.au/2017/09/17/pyscada-on-raspberry-pi-for-temperature-monitoring-with-ds18b20-on-1-wire-part-2-ds18b20-hardware-and-software/

https://cwraig.id.au/2017/09/21/pyscada-on-raspberry-pi-for-temperature-monitoring-with-ds18b20-on-1-wire-part-3-pyscada-basic-configuration/

https://cwraig.id.au/2017/09/24/pyscada-on-raspberry-pi-for-temperature-monitoring-with-ds18b20-on-1-wire-part-4-pyscada-hmi-configuration/

apanasara, 2018-02-25:
Waiting for next tutorial on modbus RTU, with solution of Dynamic Custom HMI Panel so that process/plant simulation can be displayed on HMI...

trombastic, 2018-04-17:
Dear @cwraig, many thanks for your work, i like the way you explained the setup and how seting up the HMI is working. I'd like to add the links to your Blog posts to the readme.md file in the next release and also to the documentation.
regards,
Martin

closed 2018-04-17 by trombastic