pyscada/PyScada #9: Error 500 while creating visa device

clavay, 2018-04-03:
Hi !
I'm new with Github and using PyScada. (with pyvisa(-py) and pyusb)
I don't know if it's the place to show what I found and how I resolved it.

I'm using 0.7.0rc1.
I change in visa/model.py : 
`    if type(instance) is VISADevice:
        post_save.send_robust(sender=Device, instance=instance.modbus_device)
    elif type(instance) is VISAVariable:
post_save.send_robust(sender=Variable, instance=instance.modbus_variable)`

to : 
`    if type(instance) is VISADevice:
        post_save.send_robust(sender=Device, instance=instance.visa_device)
    elif type(instance) is VISAVariable:
post_save.send_robust(sender=Variable, instance=instance.visa_variable)`

I restarted gunicorn and pyscada and I could create a visa device.

Thanks for the great job.
C.

trombastic, 2018-04-17:
Dear @clavay,

thank you very much for sharing this information with me, i will fix that in the next release. 

I saw that you added some new visa devices in your local branch, does everything it work? when you like you could add a pull request, so that i can include your devices to the project.

best wishes,
Martin

closed 2018-04-17 by trombastic

trombastic, 2018-05-11:
Dear Camille,

a short response to your questions:

> I don't see how to read string from an instrument. (*IDN for example with visa) Is it possible ?

there is a `VariableProperty` Model in `pyscada.models`, the intention for this Model was to have a place to store additional data for a variable that don't fit in the `RecordedData` or the meta data in the `Variable` Model. Maybe you could store the strings there.
```
from pyscada.models import VariableProperty
from pyscada.utils import datetime_now
from datetime import datetime

vp = VariableProperty(
             variable = var,
             property_class='data_record',
             value_string=string_data,
             timestamp=datetime_now())
vp.save()
```
> How to activate the debug in JS and where is the debug ? 

in Chrome and firefox (under Linux at least): right-click an then inspect, or shift+ctrl+c
> There is always the "zoom on x axes", "zoom on y axes", "+2h", and timeline even is there is not any chart. Is it necessary ?

this feature is relatively new, in the use-cases we using PyScada at the moment there was no need to hide the _timeline-bar_ but it should be easy to add, maybe under options the default value could be than changed in the View admin form.

Best wishes,
Martin