pyscada/PyScada #10: What is the status of this project

goanpeca, 2018-04-10:
Hi @trombastic, pretty cool project. I was wondering what is the state of the project, the plans to maintain it and keep improving it.

Cheers

trombastic, 2018-04-17:
Dear Gonzalo,

thank you for your nice words and intensest in the project, PyScada is my part time project, i am working approx 2 to 8 hours a week on the project to improve it and add new features, i have no plans to change that in the near future. I am a research fellow at the Technische Universität Berlin, we use PyScada in our lab for six different test rigs and for a field test project with 20 installations in Germany and four installations in Jordan to monitor HVAC components. Also a manufacturer of Chillers in the kW range is using PyScada to monitor his equipment.

What is Working:
 - Modbus TCP/RTU/BIN
 - Visa (at least for the Devices in the visa/devices folder)
 - systemstat
 - onewire (only DS18B20)
 - phant (no known issues)
 - smbus (at least for the Devices in the smbus/device_templates folder)
 - event (no known issues)
 - export (no known issues)
 - hmi (no known issues)

What is not Working/Missing:
 - Documentation
 - SysV init daemon handling
 - BACNet (due to the lack of hardware to test)

best wishes
Martin

goanpeca, 2018-04-17:
Great to hear that :-), no time at the moment, but will start working with this soon, hopefully I can contribute!

closed 2018-04-17 by trombastic