pyscada/PyScada #14: Is is possible to talk using IEC62056 or dlms?

bartlomieju, 2018-08-30:
Hi! 

Disclaimer: I am novice to SCADA systems. 

I would like to use PyScada to talk to energy meters using TCP. They communicate via IEC62056 and dlms protocols - I have heard that I should be able to use MODBUS to talk to them. 

Can you point me in right direction how I can do that?

trombastic, 2018-09-13:
Dear bartlomieju,

the COSEM/DLMS Protocol is not implemented in PyScada right now, there seem's to be an implementation for python, see ( [PyDLMS](https://github.com/bsdphk/PyDLMS) and  [python-dlms](https://github.com/murksermatt/python-dlms)) but i'm not sure how much of the standard they cover. There are some energy meters on the market that speak Modbus [IP, RTU]. But that should be specified in the documentation. COSEM/DLMS and Modbus are two totally different Protocols, so using Modbus to Talk to a COSEM/DLMS only device is not possible. If you have access to the protocol specification of COSEM/DLMS you could write your own plugin for PyScada.

king regards,
Martin

bartlomieju, 2018-09-22:
Thank you for information.

closed 2018-09-22 by bartlomieju