pyscada/PyScada #15: Pyscada upgrade process within the 0.7.x

cwraig, 2018-10-06:
Can you provide a command list for version upgrades within the version 0.7x stream
I am currently on PyScada Version: 0.7.0b18 and would like to move to the latest. 

I would be happy to write these up if you could provide a brief command list. 

Thanks again for the project.

trombastic, 2018-11-19:
Dear cwraig,

first of all **Make a Backup** !!!
i can not guarantee that i haven not missed a step:

Systemd
-----------

- stop all units
- remove all unit files for pyscada (`pyscada_daq`, `pyscada_export`, `pyscada_event`, `pyscada_mail`, `pyscada_modbus`, `pyscada_systemstat`)
- add and enable the new pyscada_daemon:
```
wget https://raw.githubusercontent.com/trombastic/PyScada/dev/0.7.x/extras/service/systemd/pyscada_daemon.service -O /etc/systemd/system/pyscada_daemon.service
systemctl enable pyscada_daemon
```

Django Settings
--------------------

add the logging configuration to the `settings.py`:
```
LOGGING = {
        'version': 1,
        'disable_existing_loggers': False,
        'formatters': {
            'standard': {
                'format' : "[%(asctime)s] %(levelname)s [%(name)s:%(lineno)s] %(message)s",
                'datefmt' : "%d/%b/%Y %H:%M:%S"
            },
        },
        'handlers': {
            'file': {
                'level': 'DEBUG',
                'class': 'logging.FileHandler',
                'filename': BASE_DIR + '/pyscada_debug.log',
                'formatter': 'standard',
            },
        },
        'loggers': {
            'django': {
                'handlers': ['file'],
                'level': 'INFO',
                'propagate': True,
            },
            'pyscada': {
                'handlers': ['file'],
                'level': 'DEBUG',
                'propagate': True,
            },
        },
    }
```

Upgrade
-----------

```
sudo pip install https://github.com/trombastic/PyScada/archive/dev/0.7.x.zip --upgrade
cd /var/www/pyscada/PyScadaServer
sudo -u pyscada ./manage.py migrate
sudo -u pyscada ./manage.py collectstatic
sudo -u pyscada ./manage.py pyscada_daemon init
sudo systemctl start pyscada_daemon
sudo systemctl restart gunicorn
```

that should be it,

best wishes,
Martin

apanasara, 2020-01-14:
Pyscada_daemon not starting

[image]

clavay, 2022-05-09:
Do you have still an issue ?
If not, you can close it.

clavay, 2022-06-09:
Closed due to inactivity.
Reopen if needed.

closed 2022-06-09 by clavay