pyscada/PyScada #19: Installation error "Couldn't import Django"

timjbro, 2019-05-17:
Hi @trombastic ,

Thank you for your work on this excellent project. I am looking to use PyScada as part of a range of educational equipment which my company is developing (https://www.tecquipment.com/).

I have followed the Installation Guide (https://pyscada.readthedocs.io/en/dev-0.7.x/installation.html). Unfortunately when I get to the the last step (sudo systemctl start pyscada) I get the following error (output from "journalctl -xe").

```
-- Unit pyscada.service has begun starting up.
May 09 15:19:43 debian python3[10951]: Traceback (most recent call last):
May 09 15:19:43 debian python3[10951]:   File "/var/www/pyscada/PyScadaServer/manage.py", line 8, in <module>
May 09 15:19:43 debian python3[10951]:     from django.core.management import execute_from_command_line
May 09 15:19:43 debian python3[10951]: ImportError: No module named 'django'
May 09 15:19:43 debian python3[10951]: During handling of the above exception, another exception occurred:
May 09 15:19:43 debian python3[10951]: Traceback (most recent call last):
May 09 15:19:43 debian python3[10951]:   File "/var/www/pyscada/PyScadaServer/manage.py", line 14, in <module>
May 09 15:19:43 debian python3[10951]:     import django
May 09 15:19:43 debian python3[10951]: ImportError: No module named 'django'
May 09 15:19:43 debian python3[10951]: During handling of the above exception, another exception occurred:
May 09 15:19:43 debian python3[10951]: Traceback (most recent call last):
May 09 15:19:43 debian python3[10951]:   File "/var/www/pyscada/PyScadaServer/manage.py", line 17, in <module>
May 09 15:19:43 debian python3[10951]:     "Couldn't import Django. Are you sure it's installed and "
May 09 15:19:43 debian python3[10951]: ImportError: Couldn't import Django. Are you sure it's installed and available on your PYTHONPATH environment variable? Did you forg
May 09 15:19:43 debian systemd[1]: pyscada.service: Control process exited, code=exited status=1
```

However, when I quiry the django version from the Python command line it imports and shows a version...

```
python
Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import django
>>> django.VERSION
(1, 11, 20, u'final', 0)
>>> 
```

I have also tried to follow this tutorial and @cwraig's tutorial on a Raspberry Pi 3 B+ on "Raspbian Stretch with desktop" (Release: 2019-04-08) and come to the same error.

I am new to django. Is there something obvious I've missed? What can I do to further diagnose the issue?

Thank you.
Tim

bVdCreations, 2019-05-19:
I believe you have install Django on python version 2.7
but you are trying to run the project in python3

timjbro, 2019-05-20:
Thank you. So obvious, can't believe I didn't see it! 🤦‍♂️

closed 2019-05-20 by timjbro

paulalting, 2019-05-23:
Sorry to open this again, but I have run into the very same issue, with the exact same error.
Running on Debian 9 on a ARM Cubieboard II (A20), I installed as per instructions, initially looking at setup by [https://cwraig.id.au/2017/09/17/pyscada-on-raspberry-pi-for-temperature-monitoring-with-ds18b20-on-1-wire-part-1-software-installation/](url), but noticing some installation steps were missing, like creating database, so reverted to Martin's instructions.

When I head into python (default), the version is 2.7.13, and if I list installed modules via help(modules') I see django as well as numpy and cython and pyModbus. So, it seems things are there.

If I head into python3, ver 3.5.3 and again list installed modules I don't see any of the above.

So, where do I go wrong, if I simply execute the commands as detailed by Martin ?

Martin, this looks like a great project, many thanks for your work.
Looking to set this up to talk to my solar controller over Modbus RTU and generate some better web based GUIs over my existing setup. And if all goes well, for a small industrial application.

Paul Alting (Tasmania, Australia)

timjbro, 2019-05-23:
No worries, I'm glad I'm not the only one. 😋

This may not be the best solution, but I didn't want to deal with migrating to python3.
What I did was modify the systemctl service file to force the pyscada service to run in python2.7 instead of the default python3...

```
sudo systemctl edit --full pyscada
```

Change the following lines to specify python2.7
```
ExecStart=/usr/bin/python2.7 /var/www/pyscada/PyScadaServer/manage.py pyscada_daemon start
ExecStop=/usr/bin/python2.7 /var/www/pyscada/PyScadaServer/manage.py pyscada_daemon stop
```

I don't know if you'll have the same problem, but I also found that the Gunicorn installation directory has changed from `/usr/local/bin/gunicorn` to `/usr/bin/gunicorn`.

So I modified the service file for Gunicon too...

```
sudo systemctl edit --full gunicorn
```

Hope this helps.
Tim

paulalting, 2019-05-23:
Thanks Tim for your reply. I did just do this some minutes ago and that progresses things along.
Yes, after reading your message about gunicorn, I checked, and yes, it is not in /usr/local/bin, but as you say /usr/bin, so have edited that file also and shut all applications down and do what I don't often do on this board, shutdown and reboot :)
Going to the IP address now brings up a login screen so all good there, except it doesn't appear to accept the username and password I setup for pyscada.

Martin, maybe there are some edits for the installation ?

Thanks again Tim, much appreciate your help.
I guess we can close this issue.

Paul