pyscada/PyScada #20: Read the Docs appears not to be linked to the Github repository

timjbro, 2019-05-23:
Hi @trombastic ,

Hopefully this only affects the links at the top of the Read the Docs PyScada page. However, I suspect it might be more than this, as the GitHub docs appear to be out of sync with the Read the Docs page.

Going to https://pyscada.readthedocs.io/en/dev-0.7.x/ and clicking on the "Edit on GitHub" link tries to load https://github.com/trombastic/PyScada/blob/dev-0.7.x/docs/index.rst which does not exist. It should be https://github.com/trombastic/PyScada/blob/dev/0.7.x/docs/index.rst .
Similarly going to https://pyscada.readthedocs.io/en/latest/ and clicking on the "Edit on GitHub" link tries to load https://github.com/trombastic/PyScada/blob/master/docs/index.rst . I'm not sure what this is supposed to load because the project does not contain a single master or equivalent branch.

Thank you for your help. 😊
Tim

trombastic, 2019-05-27:
Dear Tim,

thank you very much for the hint, i'm trying to fix this.

best wishes,
Martin

trombastic, 2019-06-25:
the problem is solved now, a old unused .gitmodules was the problem,
[see issue 5843 at readthedocs](https://github.com/rtfd/readthedocs.org/issues/5843)

closed 2019-06-25 by trombastic