pyscada/PyScada #23: New browser tabs created

paulalting, 2019-05-26:
When at the main site where one can select either the View pages or the Admin section, a new browser tab is created for either selection.

Also, if i click the PyScada menu item at top left corner, it also creates and moves to a new browser tab.

This should not be the case, all operation should remain within the one browser tab unless the user selects otherwise.

Very quickly, one can have a bunch of open tabs created by PyScada, which annoying as one needs to close them and just work with the one or maybe two, say one for PyScada View and one for PyScada Admin.

trombastic, 2019-05-27:
Dear Paul,

thank you very much for your opinion an that topic, i choose this behaviour because it was requested by the users here at my University, when you have more than one View and you want to open more than one it is easier to keep the view_overview in a separate tab. But maybe it is useful to have the option to change this behaviour either from the back-end (e.g. for each View) or to change it for the whole project in the `settings.py`, what is your opinion on that?

regards,
Martin

paulalting, 2019-05-27:
Having a setting in setting.py sounds best to me. Normally if I wish to open other tabs I will right click and 'open in another tab', which is easy to do.
Best,
Paul

trombastic, 2019-05-27:
ok thank you for the feedback, i will add that option in the next release.

closed 2019-06-25 by trombastic