pyscada/PyScada #24: Remove staic image from main page for Admin selection

paulalting, 2019-05-26:
On the main page where one can select the application views as well as Admin functions, there is a static image loaded under the Admin title that should be removed.
It shows details for Martin.

I went into the directory and renamed this file so it no longer loads, but this is not the best solution, as it still tries to load the image and fails with broken link reference, but does look better.

I don't see any reason for an image to be shown for Admin menu selection.

trombastic, 2019-05-27:
Dear Paul,

you are right regarding the objection that the current image shows details on my test setup, i choose to have a image there because i thought it is easier for the user to distinct between the different views and the admin link, but maybe it is better to have either no image ore a more general one. I will think about it and change it in the next release.

best wishes,
Martin

paulalting, 2019-05-27:
Maybe if a user has Admin privilege then the Admin link can be placed in the user menu as first item above user email, what do you think ?

trombastic, 2019-06-25:
yes that is a good place, i will change it that way

closed 2019-06-25 by trombastic