pyscada/PyScada #25: XY Chart: unable to have X axis set to time

paulalting, 2019-05-27:
Setting up the chart option I can set the variables for the Y axis, but there is no option for setting the X axis to the timestamp of recorded data.
How is this possible ?

[image]

paulalting, 2019-05-27:
In my SCADA system I use amCharts which, for me, appear far superior and allow far greater options for setting up and a better view for users, as an example from my system:

Is it possible to have PyScada implement the sort of functionality amCharts provides easily ?
For a SCADA system, historical charts that can be customised with a rich set of features is really important for the type of systems I do.

[image]

clavay, 2019-05-27:
I think you have to create a chart and not a xy chart.
Or do you want to use a functionality of xy chart that doesn't appear in chart ?

paulalting, 2019-05-27:
Thanks clavay, my error, I did not even notice 'charts' at the top of the HMI list.
Might be good to group like functions together rather than in alpha order.
I still find the order of views,pages,widgets and so on difficult to remember. If they were in the hierarchical order that flows and aids page design, that would be a great help.

Then place separators between  the various selections within the section to highlight it would again make things easier.

I have now created a standard chart and can view that, but still, it looks quite limited in terms of sizing and features I, and I imagine other would really need and like to have.

Thanks for your help :)

closed 2019-05-28 by paulalting