pyscada/PyScada #26: Stopping Polling of Modbus Device

paulalting, 2019-05-27:
In the Modbus Device a check box to set the device to active or not is available.
Saving a device with the 'Active' set false does not stop polling of the device as the background task that does the actual polling remains to exist and continues to operate.

The only way to stop PyScada Modbus Device from polling a slave unit is to set the device 'active' to false and save and then in 'Background Process' either delete or disable the task associated to the Modbus Device.

This task should be ended when the user set the Modbus Device 'Active' flag to false, and restarted when set to true.

clavay, 2019-05-28:
Request_data in modbus/device.py is not looking if device is set to active or not.
It's only looking if the variable is set to active.
Maybe we can add the check for active device too.

paulalting, 2019-05-28:
Yes, that would be good as it would then be possible to shutdown a complete device if required or due to critical error. It helps lower processing load and also logging errors.

Most SCADA systems allow you enable or disable a communications device.

trombastic, 2019-06-25:
dear paul,

thank you for pointing that out, i hope i solved this with the next commit

closed 2019-06-25 by trombastic