pyscada/PyScada #27: Using Custom HTML Panel

paulalting, 2019-05-28:
I have tried various methods to get variable data to be displayed in a custom HTML panel, mostly without success. What are the exact options for using this component as there is nothing in the documentation and there must have been some changes since 'cwraig's blog on it' ?

For example, if I want to use four variable which are selected from the variable list, what are the options for use in the custom HTML section, as I have tried using the variable names that I have selected, all without success ?

Also, if I need to use javascript library, say for some for of gauge or dial, can I place the included javascript here or does it need to live at some other location that will be loaded up on application load ?

Many questions :)

[image]

clavay, 2019-05-28:
To get the value of a var you can put this if the id of the var is 18 : <span class="type-numeric var-18">0.0</span>

For a variable property, it's not possible currently.

paulalting, 2019-05-28:
Yes, that is the only way I could get data to show. It is quite a pain to do it that way as you would understand. Imagine having 100 to 200 variable tags. It means paper and pencils :( 
I found that by looking inside web developer mode at how a display item is done.

Is there any chance to have my own front end and access PyScada server side ?
I have looked at the PyScada javascript and I am having trouble making any sense of it.
For my own Quadlog SCADA system I use a javascript framework which keeps all the client side a lot more readable and tidy. I use 'Backbone' with 'Underscore' and use templates to generate a 'single page application', again using Bootstrap and others.

With custom HTML, why then do you need to select variables to use if you can not then see the variable ID. It seems awkward and difficult.

clavay, 2019-05-29:
I add the options to use variables properties here : https://github.com/clavay/PyScada/commit/ca1553a1941182b31b7b9728ddbd1ad140be68a1

Maybe the way to insert a variable should be easier and more explicit.

For the javascript maybe it would be better to separate it in various files.

paulalting, 2019-05-29:
Yes, being able to insert configured variable tags for use in custom HTML components should (must) be better implemented. If it were, it would really make a difference as this should be the strong point of PyScada, by allowing the end user design and use their own javascript and HTML to present the web app in the way they need.

For me, the control items are not really usable, okay for testing, but not for a real SCADA system, where the need to design everything about the layout. For example, display couleur and size and font.

For example, for a very simple layout, I would need to be able to to the following in PyScada as I do in my own system. Is this possible in PyScada ?
[image]

Rather than something like this:
[image]
Which one would you prefer ?

I am not trying to put down PyScada, rather hoping to enthuse and encourage you to go forward and make it better :)

trombastic, 2019-06-25:
Dear Paul,

it would be great to have a gui like that in PyScada, i think with some JavaScript and HTML this should be possible also with PyScada, but at the moment i have no time for new features.  If you like you can share the code for your example with us than I will have a look on that maybe later.

If you like to do it your own beside the custom HTML Panel it is possible to write your own widgets for PyScada too.

paulalting, 2019-06-25:
Hi Martin,
I will try to spend some time looking more into the workings of PyScada to make it do more.
There are a few things that I think can be improved.
I would like to remove the use of Flot and have it agnostic about charting library, as I think there are much better charting libraries to use, such as amCharts or RGraph.
I would like to be able to make use of up to date Bootstrap, rather than what is used in PyScada.
Also, improvements with GUI responsiveness for different screen sizes. I notice what I have is okay on large screens but pretty awful on mobile screens with bad rendering.

A major concern for me is that the same database that is used for system configuration is also used to store the SCADA data from devices. The data really needs to live in its own database, separate from the configuration database. Having some methods in PyScada to manage SCADA data is also important, as at the moment, device data is stored and grows the single SqLite database. Users need a way to be able to remove or archive old data easily.

trombastic, 2019-06-26:
Hi Paul,

first of all thank you for your feedback, i really appreciate it, will respond just quick to your suggestions.
 - Replace Flot: amCharts is Commercial so for me it is out, RGraph looks interesting.
 - upgrade Bootstrap: it is on my ToDo list for very long but in my opinion there are more important tasks right now (e.g. overall stability, BACnet)
 - having a more responsive GUI that is "usable" also on mobile devices would also be nice, but it will take time that i don't have at the moment
 - using the same database for the configuration and the data is necessary because the tables are linked together and i'm pretty sure that also the big brand SCADA software use a similar architecture
 - using SQLite as DB is not recommended, with mysql or mariadb the size of the DB is not a problem, i have production systems that have databases that are >500GB and work without an issue.  
 - to archive data you can use the export, the HDF5 format is quite recourse friendly and also includes metadata from the DB 
 - adding an option for an expiration  period for the data is on my ToDo list

and last but not least, there is a developer channel on matrix.org `#pyscada:matrix.org` if you like to discus more about pyscada it is maybe better to communicate this way.

regards,
Martin

courtwarr, 2021-11-19:
Has there been any progress on this topic in since 2019?

clavay, 2021-11-22:
I am adding some things in my fork (clavay) as gauge.

I also use the PyScada db with grafana.
Look at https://github.com/clavay/PyScada/blob/master/docs/grafana.rst and https://github.com/clavay/PyScada/blob/master/extras/Grafana-test-dashboard.json

paulalting, 2021-11-22:
I went away from PySCADA for many reasons and created my own, called QuadlogSCADA.
I wanted something lightweight and fast and more adaptable for my needs in industrial control systems.
So, I made QuadlogSCADA based around two Python applications, one as the PLC or device scanner and the other as the http/s sever based on Sanic. The two apps communicate via Redis, again, lightweight and fast.
Still a lot of development work yet to do, but it is operating on an industrial site very successfully with the operator GUI being a javascript framework with standard HTML5 with all the other usual bits for a good user interface.

clavay, 2022-05-09:
I'm working on the integration of other JS library.
Reopen the issue if needed.

closed 2022-05-09 by clavay