pyscada/PyScada #28: Variables: empty COV field causes Server Error (500)

paulalting, 2019-05-29:
Having no value in the COV field for variables and saving the form causes a server error (500).
It should return with form error that field needs a value.

clavay, 2019-05-29:
I think we should replace in pyscada/models.py :
    cov_increment = models.FloatField(default=0, blank=True)
by
    cov_increment = models.FloatField(default=0)
Martin ?
Having only blank=True without null=True cause IntegrityError.

paulalting, 2019-05-29:
Also, change field title from Cov to COV as it is an acronym, for Change Of Value, or for boolean it is COS, Change Of State.

Also @trombastic, on Admin section for 'Home › PyScada Core › Variable propertys', correct spelling of propertys to properties. In English, when a word ending in 'y' become plural, it become 'ies', as in the word 'penny', becomes 'pennies'. Oui, 'Anglais' is a weird language. :)

closed 2019-06-25 by trombastic