pyscada/PyScada #29: Confusion regarding Core:Events and Mail

paulalting, 2019-06-24:
In PySCADA core events we have the ability to act on a specific event by logging the even and or sending an email. That is great, but, there is an option to select the action where there are four available selections. The option 'record and send main only when event occurs' and 'record and send mail', what is the difference ?

Also, how is it possible to specify the particular email to be sent out ?
In the Mails section, I can add different mails, each with their own subject and text body, but again, how is it possible to make these meaningful by placing variables in the main body ?

For example, If I have two mails, one for a low alarm and one for a high alarm condition, yes, I can say this in the subject, but how can I place actual variable data in the mail body ?

Then, how is it possible to then have the even select the appropriate mail based on the event that occurred ? For example, how can PySCADA send out the mail for low alarm when the event for low alarm condition is triggered, there is nowhere to specify this ?

I have configured setting.py for correct mail settings and this functions fine with a test email.

trombastic, 2019-06-25:
Dear Paul,

Let's assume the following settings for a _Event_:
Fixed limit: 20 
Limit type: value is less than limit
hysteresis: 1
Because there is a hysteresis we have two events for every _Event_, one when the actual value is above our limit + hysteresis and one when the actual value is below our limit - hysteresis.

Now you can decide what should happen PyScada can send an eMail when the event occurs  (value is above our limit + hysteresis) or when the value leaves the triggered state (value is below limit - hysteresis), _record and send mail_ will send a mail in both cases, _record and send mail only when event occurs_ will only send an eMail in the first case.

The last option you have is the level, the selection here will only change the subject text of the eMail that is send by adding the appropriate word. If you setup an _Event_ PyScada will generate an eMail and send it to all selected recipients, when the _Event_ is triggered. At the moment there is no way of changing the Text (e.g. with a Template) for that eMail. 

The Mail section in admin is just for administration to see how many an which kind of eMails have been send by pyscada.

If sending eMails with your own template is crucial for you, you can ether hack it under pyscada.models.Event -->compose_mail, or write a script for the scripting plug-in.

paulalting, 2019-06-25:
Hi Martin,
Thank you for the explanation, that makes better sense to me now.
It would be good to add this into the documentation.

closed 2019-06-25 by paulalting