pyscada/PyScada #30: Mysql issue

michaelz152, 2019-08-20:
Hi,

i have managed to install the softawre on a raspberry pi3 b+ by mixing the official installation document and cwraig notice. modbus tcp-ip is working fine so thanks a lot
i have a problem if i want to add mysql database i have the following message : 
 django.db.utils.ProgrammingError: (1146, "Table 'PyScada_db.pyscada_backgroundprocess' doesn't exist")
is someone have the same issue, or if someone have managed to install it ?

thanks

michael

clavay, 2019-08-24:
Hi,

I use Mysql.
Did you do the migrations ?
What version of PyScada do you have installed ?

clavay

michaelz152, 2019-08-26:
hi,

i use the version 0.7rc15 of piscada
i don't remember if i have made the migration because i have made a lot of thing to manage to instal piscada
and you what is your version?

thanks and good day

michael

clavay, 2019-08-28:
Hello, 

I use the last commited version, but your version should work.
Please try to do the migrations and tell me if everything is ok :
`cd /var/www/pyscada/PyScadaServer # linux`
`sudo -u pyscada python3 manage.py migrate`

clavay

clavay, 2020-10-16:
I think you miss this :
 sudo -u pyscada python3 manage.py pyscada_daemon init
Have you still an issue ?

michaelz152, 2020-10-20:
hi, 

thanks for your reply 

good day 

michael

clavay, 2022-05-09:
Reopen the issue if needed.

closed 2022-05-09 by clavay