pyscada/PyScada #31: Add FINS/UDP functionality for Omron PLCs?

rickyelopez, 2019-08-20:
I already have Python code that allows me to communicate with NX series (and probably others) PLCs over Ethernet/IP (but really FINS/UDP if I understand correctly).

With that code, how much work do you think it would be to add that functionality here? I started looking around your Modbus integration, but I'm not sure how much of it would be required for FINS because the protocol seems to be simpler.

I can provide more detail obviously just not sure what you need. If you can give me a direction to start in, I might be able to take a stab at implementing it myself and making a PR.

Thanks!

clavay, 2019-08-28:
Hello,

Could you provide your code with explanation to see how it works ?
After this we can see how to integrate it.
Maybe you can look at the visa code which is simpler.

Did you use this : https://bitbucket.org/vladimirek/omronlib/src/master/plcPyLib/
or this : https://bitbucket.org/jryan/finsgit/src/master/fins/

Cheers,
clavay

rickyelopez, 2019-09-04:
Hello,

The code I am using is a modification of https://bitbucket.org/jryan/finsgit/src/master/fins/ done by Stefano Fraccaro. Unfortunately, the site his code is on has since been taken down, however, I've attached the python file here, along with a test file that I know works for my Omron NX1P2.

I haven't looked at the VISA code, I've been basing my modifications on the Modbus code, however, I'll take a look.

Any suggestions you can provide would be very helpful, thanks!

[fins_udp.zip](https://github.com/trombastic/PyScada/files/3575052/fins_udp.zip)

clavay, 2019-09-06:
Hi,

You can use the scripting plugin to easily execute your script.

If you want to create your own plugin for this protocol, I think you can copy the visa part, defining the model, the admin part and look how device.py use the handlers which are in the devices folder and put your script and functions in one handler for your model.

@trombastic what do you think ?

rickyelopez, 2019-09-11:
Hello,

Thank you, I will look into both avenues and update this issue when I have results.

Thanks!

clavay, 2020-10-16:
Did you try it ?

rickyelopez, 2020-10-17:
@clavay haha the internship for which I was working on this ended over a year ago. I don't think we ended up pursuing this due to unrelated issues, though.

closed 2020-10-17 by rickyelopez