pyscada/PyScada #34: Pyscada Django Issue

Captain-Kovu, 2019-11-15:
I couldn't seem to complete the installation process, could someone help me? I haven't got a clue what's wrong. I could still get the django server up and running, but the issue is I only get text on the screen (see picture below). Any help will be greatly appreciated, I need to set this up for my course in University. Many thanks!

Currently using python 3.7.5

D:\Aqiel\Python Project\WindTurbineScada>python manage.py pyscada_daemon --init
Traceback (most recent call last):
  File "manage.py", line 22, in <module>
    execute_from_command_line(sys.argv)
  File "C:\Users\mirza\AppData\Local\Programs\Python\Python37\lib\site-packages\django\core\management\__init__.py", line 364, in execute_from_command_line
    utility.execute()
  File "C:\Users\mirza\AppData\Local\Programs\Python\Python37\lib\site-packages\django\core\management\__init__.py", line 356, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "C:\Users\mirza\AppData\Local\Programs\Python\Python37\lib\site-packages\django\core\management\__init__.py", line 206, in fetch_command
    klass = load_command_class(app_name, subcommand)
  File "C:\Users\mirza\AppData\Local\Programs\Python\Python37\lib\site-packages\django\core\management\__init__.py", line 40, in load_command_class
    module = import_module('%s.management.commands.%s' % (app_name, name))
  File "C:\Users\mirza\AppData\Local\Programs\Python\Python37\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Users\mirza\AppData\Local\Programs\Python\Python37\lib\site-packages\pyscada\management\commands\pyscada_daemon.py", line 6, in <module>
    from pyscada.utils.scheduler import Scheduler
  File "C:\Users\mirza\AppData\Local\Programs\Python\Python37\lib\site-packages\pyscada\utils\scheduler.py", line 30, in <module>
    from os import linesep, umask, access, W_OK, kill, remove, setsid, path, fork, F_OK, WNOHANG, getpid, waitpid
ImportError: cannot import name 'setsid' from 'os' (C:\Users\mirza\AppData\Local\Programs\Python\Python37\lib\os.py)

[image]

clavay, 2019-11-15:
Try running it on unix, os.setsid is not available on Windows and I don't know how to replace this.

Maybe @trombastic know...

Captain-Kovu, 2019-11-21:
So I restarted and followed the instructions on the documentation in Ubuntu.. 
Installation ran smoothly but the site remains the same, no graphics whatsoever

clavay, 2019-11-21:
Did you do the collectstatic line during the installation ?
`sudo -u pyscada python3 manage.py collectstatic`

Captain-Kovu, 2019-11-21:
Yes I have

clavay, 2019-11-22:
can you put the output of the collectstatic command ?
Try to refresh the page with Ctrl+F5.

Captain-Kovu, 2019-11-23:
$ sudo python3 manage.py collectstatic

You have requested to collect static files at the destination
location as specified in your settings:

    /var/www/pyscada/http/static

This will overwrite existing files!
Are you sure you want to do this?

Type 'yes' to continue, or 'no' to cancel: yes

0 static files copied to '/var/www/pyscada/http/static', 178 unmodified.

clavay, 2019-11-24:
Try to restart gunicorn : sudo systemctl restart gunicorn
And then refresh the page in your browser with : ctrl + f5

If not look if you have any message in the console of the browser or set the log level to debug for gunicorn and post the logs.

clavay, 2020-10-16:
If it's ok you can close the issue

clavay, 2022-05-09:
Reopen the issue if needed.

gsnitz, 2022-10-25:
I'm having the same issue described above. 

The output of `collectstatic` is the same as Captain-Kovu's. 

Restarting gunicorn and hard browser refresh did not product any change. 

Running on Raspberry-Pi-Desktop-OS (Debian) inside a VM. 

[image]

clavay, 2022-10-26:
Hello,
i just saw that Captain-Kovu access to Gunicorn through the 8000 port which is not good because I is not using Nginx which indicate where are the static files.

Can you send me the content of these files :
/etc/nginx/sites-enabled/pyscada.conf
/etc/systemd/system/gunicorn.service
/etc/systemd/system/gunicorn.socket
/var/www/pyscada/PyScadaServer/PyScadaServer/settings.py  (remove the secret key before copying)

gsnitz, 2022-10-26:
### FILE DUMP

/etc/nginx/sites-enabled/pyscada.conf
```
# pyscada.conf

# the upstream component nginx needs to connect to
upstream app_server {
        server unix:/tmp/gunicorn.sock fail_timeout=0; # for a file socket
        #server 127.0.0.1:8000 fail_timeout=0; # for a file socket
}

# configuration of the server
server {
    listen      80;
    listen   [::]:80;
    server_name _;          # substitute your machine's IP address or FQDN    **<<-- COULD BE SOURCE OF ISSUE**  
    #return 301 https://$server_name$request_uri;
    return 301 https://$host$request_uri;
}

server {
        listen   443 default_server ssl;
        listen [::]:443 ssl;

        server_name _;          # substitute your machine's IP address or FQDN    **<<-- COULD BE SOURCE OF ISSUE**  

        charset utf-8;
        keepalive_timeout 5;
        client_max_body_size 75M;   # max upload size, adjust to taste
        # please comment if https is not used
        ssl_certificate     /etc/nginx/ssl/pyscada_server.crt; # The certificate file
        ssl_certificate_key /etc/nginx/ssl/pyscada_server.key; # The private key file

        # Django media
        location /media  {
                alias /var/www/pyscada/http/media;  # your Django project's media files - amend as required
        }

        location /static {
                alias /var/www/pyscada/http/static; # your Django project's static files - amend as required
        }

        location /measurement {
                alias /home/pyscada/measurement_data_dumps; # to support download of measurement files via admin backend - amend as required
        }

        location / {
            # checks for static file, if not found proxy to app
            try_files $uri @proxy_to_app;
        }

        location @proxy_to_app {
            proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
            proxy_set_header Host $http_host;
            proxy_redirect off;

            proxy_pass   http://app_server;
        }
}
```

/etc/systemd/system/gunicorn.service
```
[Unit]
Description=gunicorn daemon
Requires=gunicorn.socket
After=network.target

[Service]
[... 280 lines cut ...]

/var/www/pyscada/PyScadaServer/PyScadaServer/settings.py

I added all of the specified `INSTALLED_APPS` but then commented out `#'pyscada.visa',` `#'pyscada.onewire',` and `#'pyscada.smbus',` since I'm not intending on using those and didn't `pip3 install pyvisa pyvisa-py`, `apt-get install owfs`, `pip3 install pyownet`, `apt-get install libffi-dev` or `pip3 install smbus-cffi` for the required packages. 

In `DATABASES`, I didn't change the default `PyScada-user` and `PyScada-user-password` for now while I'm trying to get through basic testing. I will remember to change these later. 

Looks like  put the special file encoding header [pep-0263](https://peps.python.org/pep-0263/) in the **middle** of the file instead of in the header. I used `sudo systemctl stop pyscada` .  Then back to `nano ./settings.py` to edit the file. 

I moved these lines up to the top of the file. That's the point where I also noticed I had left the second line as `# -*- coding: <encoding name> -*-` and did not specify a specific encoding. So I edited the encoding as shown below. Then `systemctl restart pyscada` and `CTRL-F5` to hard refresh browser. 

- The first edit was to set `# -*- coding: latin-1 -*-` - could not start service.
- The second edit was to set `# -*- coding: utf-8 -*-` - service started but no change.
- The third edit was to set `# -*- coding: utf-16 -*-`  could not start service. 
- The fourth edit was to set `# -*- coding: ascii -*-` - service started but no change.

Don't see any issues with the remainder of the file. 

**File was left with this last edit in place.** 

I went ahead and rebooted the VM at this point as a final check - no change. 

### RESULTS

Didn't get very far on my own. Might have fixed an issue or two but no visible impact to webserver.

clavay, 2022-10-27:
Can you send me the version of django you have ?
```
sudo pip3 list | grep Django
```

Can you send me the log of the console from your browser : https://appuals.com/open-browser-console/

Can you send me the log of the pyscada log file : tail -n200 /var/log/pyscada_debug.log

I see you have the version 0.7.0rc15, the actual version is 0.7.1rc1, can you update to the new version and migrate, collectstatic and restart gunicorn and pyscada using systemctl ?

clavay, 2022-10-27:
Your log should be here :  
```
tail -n200 /var/www/pyscada/PyScadaServer/pyscada_debug.log
```

Upgrade with : 
```
sudo pip3 install --upgrade https://github.com/pyscada/PyScada/tarball/master
```

Then : 
```
#Migration and static files
cd /var/www/pyscada/PyScadaServer # linux
sudo -u pyscada python3 manage.py migrate
sudo -u pyscada python3 manage.py collectstatic --noinput

# load fixtures with default configuration for chart lin colors and units
sudo -u pyscada python3 manage.py loaddata color
sudo -u pyscada python3 manage.py loaddata units

# initialize the background service system of pyscada
sudo -u pyscada python3 manage.py pyscada_daemon init

# restart the services
sudo systemctl restart gunicorn
sudo systemctl restart pyscada
```

gsnitz, 2022-10-28:
### Log

```
[27/Oct/2022 16:10:01] DEBUG [pyscada.utils.scheduler:706] PID 930, received signal: 15
[27/Oct/2022 16:10:01] DEBUG [pyscada.utils.scheduler:493] try to terminate process id 930
[27/Oct/2022 16:10:01] DEBUG [pyscada.utils.scheduler:706] PID 864, received signal: 15
[27/Oct/2022 16:10:01] DEBUG [pyscada.utils.scheduler:706] PID 1037, received signal: 15
[27/Oct/2022 16:10:01] DEBUG [pyscada.utils.scheduler:706] PID 883, received signal: 15
[27/Oct/2022 16:10:01] DEBUG [pyscada.utils.scheduler:493] try to terminate process id 1037
[27/Oct/2022 16:10:02] DEBUG [pyscada.utils.scheduler:493] try to terminate process id 864
[27/Oct/2022 16:10:02] DEBUG [pyscada.utils.scheduler:493] try to terminate process id 883
[27/Oct/2022 16:10:02] DEBUG [pyscada.utils.scheduler:706] PID 864, received signal: 15
[27/Oct/2022 16:10:02] DEBUG [pyscada.utils.scheduler:493] try to terminate process id 930
[27/Oct/2022 16:10:02] DEBUG [pyscada.utils.scheduler:706] PID 883, received signal: 15
[27/Oct/2022 16:10:02] DEBUG [pyscada.utils.scheduler:493] try to terminate process id 1037
[27/Oct/2022 16:10:02] DEBUG [pyscada.utils.scheduler:706] PID 930, received signal: 15
[27/Oct/2022 16:10:02] DEBUG [pyscada.utils.scheduler:706] PID 1037, received signal: 15
[27/Oct/2022 16:10:03] DEBUG [pyscada.utils.scheduler:493] try to terminate process id 864
[27/Oct/2022 16:10:03] DEBUG [pyscada.utils.scheduler:493] try to terminate process id 883
[27/Oct/2022 16:10:03] DEBUG [pyscada.utils.scheduler:493] try to terminate process id 930
[27/Oct/2022 16:10:03] DEBUG [pyscada.utils.scheduler:493] try to terminate process id 1037
[27/Oct/2022 16:10:03] DEBUG [pyscada.utils.scheduler:706] PID 1037, received signal: 15
[27/Oct/2022 16:10:04] DEBUG [pyscada.utils.scheduler:382] process pyscada.modbus-1-192.168.20.51:502-0 started
[27/Oct/2022 16:10:04] DEBUG [pyscada.utils.scheduler:493] try to terminate process id 1037
[27/Oct/2022 16:10:06] DEBUG [pyscada.utils.scheduler:483] termination of the daemon done
[27/Oct/2022 16:10:06] DEBUG [pyscada.utils.scheduler:230] process 835 tried to delete pid
[27/Oct/2022 16:10:12] ERROR [pyscada.modbus.device:299] device with id: 1 is not accessible
[27/Oct/2022 16:11:08] INFO [pyscada.utils.scheduler:214] created pid 1641
[27/Oct/2022 16:11:08] DEBUG [pyscada.utils.scheduler:279] process 126 is alive
[27/Oct/2022 16:11:08] DEBUG [pyscada.models:1327] send sigterm to daemon
[27/Oct/2022 16:11:08] DEBUG [pyscada.utils.scheduler:706] PID 1636, received signal: 15
[27/Oct/2022 16:11:08] DEBUG [pyscada.utils.scheduler:382] process pyscada.systemstat started
[27/Oct/2022 16:11:08] DEBUG [pyscada.utils.scheduler:382] process pyscada.modbus started
[27/Oct/2022 16:11:08] DEBUG [pyscada.utils.scheduler:382] process pyscada.event started
[27/Oct/2022 16:11:08] DEBUG [pyscada.utils.scheduler:382] process pyscada.mail started
[27/Oct/2022 16:11:08] DEBUG [pyscada.utils.scheduler:382] process pyscada.export started
[27/Oct/2022 16:11:08] DEBUG [pyscada.utils.scheduler:382] process pyscada.modbus-1-192.168.20.51:502-0 started
[27/Oct/2022 16:11:10] DEBUG [pyscada.utils.scheduler:382] process pyscada.modbus-1-192.168.20.51:502-0 started
[27/Oct/2022 16:11:15] DEBUG [pyscada.utils.scheduler:394] process 126 is dead
[27/Oct/2022 16:11:16] ERROR [pyscada.modbus.device:299] device with id: 1 is not accessible
[27/Oct/2022 16:11:18] ERROR [pyscada.modbus.device:299] device with id: 1 is not accessible
[27/Oct/2022 16:11:35] DEBUG [pyscada.utils.scheduler:382] process pyscada.modbus-1-192.168.20.51:502-0 started
[27/Oct/2022 16:11:38] DEBUG [pyscada.utils.scheduler:382] process pyscada.modbus-1-192.168.20.51:502-0 started
[27/Oct/2022 16:11:48] ERROR [pyscada.modbus.device:299] device with id: 1 is not accessible
[27/Oct/2022 16:11:51] ERROR [pyscada.modbus.device:299] device with id: 1 is not accessible
[27/Oct/2022 16:24:36] DEBUG [pyscada.utils.scheduler:580] PID 1641, received signal: 12
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:573] pyscada.utils.scheduler.Scheduler, parrent process_id 1
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:574] pyscada.utils.scheduler.Scheduler, process_id 1
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:573] pyscada.utils.scheduler.Scheduler, parrent process_id 1
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:574] pyscada.utils.scheduler.Scheduler, process_id 1
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:573] pyscada.utils.scheduler.Scheduler, parrent process_id 3
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:574] pyscada.utils.scheduler.Scheduler, process_id 1
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:573] pyscada.utils.scheduler.Scheduler, parrent process_id 3
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:574] pyscada.utils.scheduler.Scheduler, process_id 1
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:573] pyscada.utils.scheduler.Scheduler, parrent process_id 1
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:574] pyscada.utils.scheduler.Scheduler, process_id 1
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:573] pyscada.utils.scheduler.Scheduler, parrent process_id 1
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:574] pyscada.utils.scheduler.Scheduler, process_id 1
[27/Oct/2022 16:24:37] DEBUG [pyscada.utils.scheduler:573] pyscada.utils.scheduler.Scheduler, parrent process_id 1
[... cut ...]
  File "/usr/lib/python3/dist-packages/MySQLdb/cursors.py", line 374, in _do_query
    db.query(q)
  File "/usr/lib/python3/dist-packages/MySQLdb/connections.py", line 292, in query
    _mysql.connection.query(self, query)
django.db.utils.IntegrityError: (1062, "Duplicate entry '3495878812386721793' for key 'PRIMARY'")
```

All other steps completed.

VM rebooted.

### Results

There's now an interactable icon for the `Data refresh rate: ms` at the home page:

[image]

No observable change to Admin page:

[image]

clavay, 2022-10-28:
I think nginx is the problem.
Can you show me the output of :
```
sudo systemctl status nginx
wget -O/dev/null http://127.0.0.1:80/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js
```

Other thing, your error is strange : 
```
File "/usr/local/lib/python3.7/dist-packages/pyscada/utils/scheduler.py", line 645, in run
    RecordedData.objects.bulk_create(item)
```

Because the line 645 of scheduler.py is not : RecordedData.objects.bulk_create(item)
it is : https://github.com/pyscada/PyScada/blob/master/pyscada/utils/scheduler.py#L645
Did you restart pyscada after upgrading ?
Can you send the log after restarting pyscada ?

gsnitz, 2022-11-03:
Apologies for the long delay, work/life have been a bit hectic.

### NGINX

`sudo systemctl status nginx`

```
● nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2022-11-03 12:10:41 CDT; 25s ago
     Docs: man:nginx(8)
  Process: 521 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
  Process: 556 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
 Main PID: 569 (nginx)
    Tasks: 3 (limit: 2194)
   Memory: 10.5M
   CGroup: /system.slice/nginx.service
           ├─569 nginx: master process /usr/sbin/nginx -g daemon on; master_process on;
           ├─577 nginx: worker process
           └─578 nginx: worker process

Nov 03 12:10:40 raspberry systemd[1]: Starting A high performance web server and a reverse proxy server...
Nov 03 12:10:41 raspberry systemd[1]: Started A high performance web server and a reverse proxy server.
```

`wget -O/dev/null http://127.0.0.1:80/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js`

```
--2022-11-03 12:12:46--  http://127.0.0.1/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js
Connecting to 127.0.0.1:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://127.0.0.1/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js [following]
--2022-11-03 12:12:46--  https://127.0.0.1/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js
Connecting to 127.0.0.1:443... connected.
ERROR: The certificate of ‘127.0.0.1’ is not trusted.
ERROR: The certificate of ‘127.0.0.1’ doesn't have a known issuer.
The certificate's owner does not match hostname ‘127.0.0.1’
```

**Perhaps I need to recreate the SSL certificate?**

Looking back, I had `server_name _; # substitute your machine's IP address or FQDN` in `/.../nginx/.../pyscada.conf` before I created the SSL, and did not recreate the certificate after changing it to the LOOPBACK address `server_name 127.0.0.1;`.

Thoughts?

### SCHEDULER

Quick excerpt from my local copy of `/usr/local/lib/python3.7/dist-packages/pyscada/utils/scheduler.py`:

[image]

### RESTART

In my previous post, I had executed the `sudo systemctl restart pyscada` and then I also rebooted the VM.

In today's post, I booted the VM to get the above information, and then executed the `sudo systemctl restart pyscada` command again:

[image]

### CURRENT STATUS
[image]

clavay, 2022-11-03:
Did this files exists? 
```
ssl_certificate     /etc/nginx/ssl/pyscada_server.crt; # The certificate file
ssl_certificate_key /etc/nginx/ssl/pyscada_server.key; # The private key file
````
You can recreate them with :
```
 sudo openssl req -x509 -nodes -days 1780 -newkey rsa:2048 -keyout /etc/nginx/ssl/pyscada_server.key -out /etc/nginx/ssl/pyscada_server.crt -subj '/CN=www.mydom.com/O=My Company Name LTD./C=US'
```
If it is not working can you replace the nginx pyscada config by this one : 
```
https://raw.githubusercontent.com/clavay/PyScada-Laborem/master/extras/nginx_sample.conf
```
and then restart nginx and try again : 
```
wget -O/dev/null http://127.0.0.1:80/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js

```

gsnitz, 2022-11-04:
They exist:

[image]

Recreated ssl - no change

Reaplced nginx pyscada config, restarted nginx, tried wget again - no change

[image]

Perhaps it would be worth testing http connection only (disable ssl)?

clavay, 2022-11-05:
Yes you can try. 
But the result of wget after changing the nginx conf is strange.
Because this nginx conf doesn't redirect http to https as the old one you have do...
Are you sure the new conf is used? 
Did you restart the VM to be sure?

gsnitz, 2022-11-10:
Although this is unresolved, go ahead and close the issue. I'll find another way to accomplish my goals.

clavay, 2023-03-23:
I made an [install](https://github.com/pyscada/PyScada/blob/master/install.sh) script and it works in a debian VM.
Reopen if you have still an issue.

closed 2023-03-23 by clavay