pyscada/PyScada #36: Failed to start PyScada Daemon

apanasara, 2020-01-14:
PySCADA Deamon is not failing to start after upgradation

[image]

clavay, 2020-01-14:
Can you put the output of journalctl -xe ?
What is your version of pyscada ?

apanasara, 2020-01-15:
@clavay , Please refer below status

[image]

apanasara, 2020-01-15:
I imported _include_ by  following into urls.py but still getting error

```python
from django.conf.urls import include
```

> **Now error is**
> 1146, "Table 'PyScada_db.pyscada_backgroundprocess' doesn't exist"

[image]

clavay, 2020-01-17:
Did you successfully do this : 
`sudo -u pyscada python3 manage.py pyscada_daemon init`

clavay, 2022-05-09:
Reopen the issue if needed.

closed 2022-05-09 by clavay