pyscada/PyScada #40: Can't start pyscada

WorkKMc, 2020-10-15:
I've been working on trying to set up pyscada. Every time I try the command "sudo systemctl start pyscada" it never works. I've gone through the documentation looking to find if I can figure out why the command isn't working and I just can't seem to figure out why. I'm using a raspberry pi 3 B+, the os is Raspbian jessie 8. I added a screenshot of what comes up when I run the following commands "sudo systemctl start pyscada", "sudo journalctl -xn", and "sudo systemctl status pyscada.service".
[image]

WorkKMc, 2020-10-15:
It doesn't look like I have a pyscada_debug.log file. Should It been downloaded or did one need to be created?

clavay, 2020-10-16:
how did you install python ?
can you show the output of : which python3

WorkKMc, 2020-10-16:
[image]

WorkKMc, 2020-10-16:
I usually use python3.6

clavay, 2020-10-16:
and the result of :
which python3
which python3.6
?

clavay, 2020-10-16:
try this : /usr/bin/python3 /var/www/pyscada/PyScadaServer/manage.py pyscada_daemon start

I miss the caps, the log file is there : tail -n200 /var/www/pyscada/PyScadaServer/pyscada_debug.log

WorkKMc, 2020-10-16:
I'm sorry the try line do you mean command line or in the pyscada.service file as the ExcStart line?

clavay, 2020-10-16:
in command line

WorkKMc, 2020-10-16:
Ok so running that in the command line I get the first picture. I do have Dajngo installed but for python 3.6.
[image]

Then I use the /home/pi/Python-3.6.5/python to run python 3.6. This is what I get running python 3.6.
[image]

clavay, 2020-10-16:
With your installation of python in /home/pi/Python-3.6.5 something is missing in python, but the reason doesn't seem to be pyscada nor django.
I don't know why you want to use this way of using python and how you install it.

With the normal installation of python (/usr/bin/python), you didn't follow the instructions (http://pyscada.rtfd.io) so you don't have django installed.

My suggestion is to follow the doc, and don't use another path for python, if you need a special version of python you could install it using : sudo apt-get install python3.6

Tell me if you need help for something during the installation process.

WorkKMc, 2020-10-16:
Ok I'll let you know if it works.

clavay, 2020-11-02:
Do you have still an issue ?

WorkKMc, 2020-11-02:
I couldn't install python 3.6 the why you suggested. Sorry for the late respond. Other projects at work have started be higher priority.

clavay, 2022-05-09:
Reopen the issue if needed.

closed 2022-05-09 by clavay