pyscada/PyScada #41: Problems using PyScada for modbus device

cyberitot79, 2021-03-10:
Hi there, 

I wanted to ask about some problems I have while I'm trying to use PyScada as an HMI for a modbus device. Let me explain the scenario to let you know what I'm trying to do. I have one Conpot honeypot (https://github.com/mushorg/conpot) running to simulate a PLC device that uses Modbus protocol. It's working fine as I can connect to it with some clients (for example mbtget) and also reading values (coils, discrete, holding... all of them). So I wanted to set an HMI just to get like a GUI where I can show some charts, tables and values that I read from that simulated PLC with Conpot. So, I took a look and I think PyScada is perfect for it.

So, I've installed it following your official documentation (the host version and also the docker version, in different machines, just to try if there was any difference). And once I've installed it and launched it, I can perfectly connect to admin GUI where I can set the scenario for the HMI. Just as a try, I've followed these steps (https://cwraig.id.au/2017/09/24/pyscada-on-raspberry-pi-for-temperature-monitoring-with-ds18b20-on-1-wire-part-4-pyscada-hmi-configuration/) where they explain how to set an easy HMI in a simple way, you know, creating charts, panels, items, view... I've also defined a Modbus TCP device and some variables, and that's not the problem as I've set the right IP address, port, registers addresses and type (coils...). 

The problem is that, whatever I define, more or less variables, more or less elements, charts... they never give me a variable value, it's like it's never reading. I can see the tables and charts on the view, that's not the problem, but they're empty. And it's not just a problem of not being reading. The thing is that I don't see any kind of interaction in my Conpot log, like, if it never got an IP connection try. 

So I'd like to ask if, besides launching the PyScada, defining the Modbus device, defining the view and HMI elements... I'd need to do something else to make it work, as I feel like it never sends a connection try.

 Just to help to understand it, I'm leaving some pictures of what I've configured. I just defined 2 variables to read holding registers as an example. About variable addresses, I used one with address 1 and another with 40001 because I'm not sure if it starts in 1 mapping to 40001, or if I have to set the 40001 number.

Thank you so much!!

[image]
[image]
[image]

clavay, 2021-03-10:
Hi,

Can you give us the logs from /var/www/pyscada/PyScadaServer/pyscada_debug.log

Did you try with FC3 or FC4 ?

cyberitot79, 2021-03-11:
Hi again!

I'm not sure why, but doing exactly the same than yesterday, now I can connect them and read all the values without a problem.
I'm sorry for bothering with the problem.

Anyways, thank you so much! For your help and for such a great project!! :D

clavay, 2021-03-11:
When you change or add a device/variable, you have to wait for pyscada to restart the process associated to the device to apply your changes.

You can close the issue if you don't have any other question for the moment.

cyberitot79, 2021-03-12:
Yeah, I didn't know that and I guess that was the problem the first day, but now it perfectly works. And sure, I'm closing it now. Sorry that I'm new at Github and I thought it had to be closed by the owner of the project. Thanks!!

closed 2021-03-12 by cyberitot79