pyscada/PyScada #52: Table pyscada_db.pyscada_deviceprotocol doesnt exist

Karsak4747, 2023-05-13:
When I install pycada according to the instructions or via Docker, I get the error django.db.utils.programmingerror: (1146, "table pyscada_db.pyscada_deviceprotocol doesnt exist"). How to get rid of it?
Installing on debian11

clavay, 2023-05-16:
Hello,
I am working on the version 0.8.0 of PyScada and I will publish it soon.
It will come with a system and docker installation script.

If I have time, I will publish it today.

Your error sounds like you didn't run the migrations.
`sudo python3 /var/www/pyscada/PyScadaServer/manage.py migrate`

Karsak4747, 2023-05-18:
Thank you very much, now the installation has worked! Please tell me how to integrate the protocols in the version with docker installation.
I'm trying to use the command: pip install pyscada-modbus
And I get the error: ERROR: Could not find a version that satisfies the requirement pyscada-modbus (from versions: none)
ERROR: No matching distribution found for pyscada-modbus

clavay, 2023-05-19:
Try `pip3 install https://github.com/pyscada/pyscada-modbus/tarball/master` as we didn't publish it in pypi the last versions.

Karsak4747, 2023-05-19:
Thank you very much!

closed 2023-05-19 by Karsak4747