pyscada/PyScada #102: Recorded Data init : value class seems not found

clavay, 2023-12-18:
from pyscada/PyScada-Modbus#2 :

```
[17/Dec/2023 09:46:29] ERROR [pyscada.utils.scheduler:927] pyscada.modbus-1-192.168.1.59:502-1, unhandled exception
Traceback (most recent call last):
  File "/home/pyscada/.venv/lib/python3.10/site-packages/pyscada/utils/scheduler.py", line 818, in run
    status, data = self.loop()
  File "/home/pyscada/.venv/lib/python3.10/site-packages/pyscada/utils/scheduler.py", line 1764, in loop
    tmp_data = device.request_data()
  File "/home/pyscada/.venv/lib/python3.10/site-packages/pyscada/modbus/device.py", line 330, in request_data
    recorded_data_element = self.variables[variable_id].create_recorded_data_element()
  File "/home/pyscada/.venv/lib/python3.10/site-packages/pyscada/models.py", line 1731, in create_recorded_data_element
    return RecordedData(
  File "/home/pyscada/.venv/lib/python3.10/site-packages/pyscada/models.py", line 2802, in __init__
    super(RecordedData, self).__init__(*args, **kwargs)
  File "/home/pyscada/.venv/lib/python3.10/site-packages/django/db/models/base.py", line 567, in __init__
    raise TypeError(
TypeError: RecordedData() got unexpected keyword arguments: 'value'
```

Kimyasallar, 2024-01-02:
Can the reason of this bug be the "get_bits_by_class" function is not having definition of UINT64?

pyscada.modbus.models.py line 1359:
def get_bits_by_class(self):
        """
        `BOOLEAN`							1	1/16 WORD
        `UINT8` `BYTE`						        8	1/2 WORD
        `INT8`								8	1/2 WORD
        `UNT16` `WORD`						16	1 WORD
        `INT16`	`INT`						16	1 WORD
        `UINT32` `DWORD`					        32	2 WORD
        `INT32`								32	2 WORD
        `FLOAT32` `REAL` `SINGLE` 			        32	2 WORD
        `FLOAT48` 'INT48'                  	                        48	3 WORD
        `FLOAT64` `LREAL` `FLOAT` `DOUBLE`	        64	4 WORD
        """

clavay, 2024-01-08:
This bug is not related to this function.

The value class choices defined in the `Variable` [model](https://github.com/pyscada/PyScada/blob/79b98d00107caa436181cb6638085743271beea4/pyscada/models.py#L1233-L1260) are : 
```
UNIXTIMEF64
UNIXTIMEI64
UNIXTIMEF32
UNIXTIMEI32
FLOAT64
FLOAT48
FLOAT32
UINT64
UINT32
UINT16
UINT8
INT64
INT48
INT32
INT16
INT8
BOOLEAN
```

But the [`__init__`](https://github.com/pyscada/PyScada/blob/79b98d00107caa436181cb6638085743271beea4/pyscada/models.py#L2752-L2799)  function of the `RecordedData` model does not know the `UINT64` value class to choose a column to store it, and does not have a default option.

I will do a PR.

clavay, 2024-01-09:
@Kimyasallar can you try the PR proposed ?

Kimyasallar, 2024-01-09:
Now will try as hardcoded @clavay

Kimyasallar, 2024-01-09:
The first feedback is:

No error is returned anymore, but no data is read either.

Will try more later today. Thank you.

Kimyasallar, 2024-01-09:
The latest feedback is:

Before update, I was reading the uint64 addresses as uint32 value_class varibles with wrong values, but I was able to read some values as it is.
Now with the update, no value is stored from uint64 addresses. It shows up None. :(

clavay, 2024-01-09:
I miss something, I update the `init` of the data in the `RecordedData` table (writing value to the db) but I didn't update the `value` function (read value from the db).
Please try with the PR #105 I just updated.

Kimyasallar, 2024-01-10:
Hi @clavay,

I have tryed the both updates and now thankfully I am getting some values. But now the new situation is with the float readings.

For UINT64 addressed variables that I am reading I have results like 1.8446744073709552e+19.
For float addressed variables that I already was reading as float, have results like 5.289724578700294e-35.

As I don't have any problems with uint32 and int32 variables, I suspect the floats. Any idea about this?

Thank you again 🙏🏼

clavay, 2024-01-11:
The result you see is done by de [decode_value](https://github.com/pyscada/PyScada/blob/2968d9f6031d5eec47aa87b026aa6f0e5ef03627/pyscada/models.py#L1487-L1627) function.

Maybe I never try if the decoding is correctly done for a UINT64, I don't remember.

For UINT64 you have :
```
target_format = "Q"
source_format = "4H"
```
These formats are specified [here](https://docs.python.org/3/library/struct.html#format-characters).

Maybe the `target_format` is wrong, try to replace it by `2Q` and look at the result.
Other option, the decoding that depend on the byte order is not done correctly for `source_format="4H"` : [see here](https://github.com/pyscada/PyScada/blob/2968d9f6031d5eec47aa87b026aa6f0e5ef03627/pyscada/models.py#L1595-L1627)

To help me finding the good values for the `target_format` and the decoding method, please add a log after line 1595 : 
`logger.info(value)`
Send me the result of the log and the expected value to read.

Kimyasallar, 2024-01-14:
With target_format = "2Q" update, I got the below error:
struct.error: unpack requires a buffer of 16 bytes

Will go on with your other recommendations later on. Thank you.

clavay, 2024-02-09:
@Kimyasallar , can you try the branch of the PR #122 ?

Kimyasallar, 2024-02-10:
Is this added to 0.8.2?

Kimyasallar, 2024-02-13:
Hi @clavay, me again :)

With the new update, I have tried to read UINT64 value. I got the below error:

ValueError: Saving value to RecordedData for 16 - Consumed_Energy_L1 with value class UINT64 should be in the interval [0:18446744073709551616], it is 18446744073709551616

Can you help me with this please?

trombastic, 2024-02-13:
(2**64)-1  = 18446744073709551615 is the maximum value a UINT64 can store so 18446744073709551616 is to big. The error message is misleading, we will fix that.

Kimyasallar, 2024-02-13:
Thank you Martin. Can you tell me the location of the error or give insight what to do for a quick solution locally?

clavay, 2024-02-13:
I propose a PR to fix this.

closed 2024-02-13 by trombastic

Kimyasallar, 2024-02-13:
I have tried. It didn't work. We made it smaller, isn't it?

ValueError: Saving value to RecordedData for 16 - Consumed_Energy_L1 with value class UINT64 should be in the interval [0:18446744073709551615], it is 18446744073709551616

trombastic, 2024-02-13:
the value is to high to be stored as a uint64, 18446744073709551616 is 1 greater than the maximum value that can be stored, so you have to choose a smaller value. I'm not sure where the problem is, what is the actual value that you try to store?

Kimyasallar, 2024-02-13:
The device says that the address is UINT64. I couldn't see the value as I can't read now. I will try with mbpoll now.

Kimyasallar, 2024-02-13:
It is weird. I have changed the address to be read on the same device with another UINT64 value and now it reads.

🤷🏻‍♂️ What to say :)

PR #125 (2024-02-13): fix high limit for UINT64
should fix #102