pyscada/PyScada #103: Various variable value class with the same key

clavay, 2023-12-18:
The value class key should be unique.
Keep only one value for : 
- float 32, 48, 64
- int 8, 16, 32, 64
- uint 8, 16, 32, 64
- boolean

Should we add other value classes ?

@trombastic : what is unixtime[i or f][32 or 64]  value class for ?

https://github.com/pyscada/PyScada/blob/79b98d00107caa436181cb6638085743271beea4/pyscada/models.py#L1233-L1260

trombastic, 2024-01-12:
>> The value class key should be unique.
proberbly, this was added because in many systems the data types have different names, this should help the user to choose the right one, but i agree with you that it adds space for confusion and should therefore be changed.

the UINT64 is missing because there is no way to store it in some of the databases, it is always is casted to IN64, this may have changed in the meantime

>> what is unixtime[i or f][32 or 64] value class for
storing a unixtimestamp as int32, int64 or float32, float64 in the recoded data database, so the user can decide if he want's  sub second resolution or more range

clavay, 2024-01-12:
> > > The value class key should be unique.

> proberbly, this was added because in many systems the data types have different names, this should help the user to choose the right one, but i agree with you that it adds space for confusion and should therefore be changed.

but when you don't choose the first key in the list, after saving the admin variable page it will display the first element matching the key.
For example choosing `("FLOAT64", "FLOAT64")` will always display after saving `("FLOAT64", "LREAL (FLOAT64)")`.
It may be confusing for the user.

-----------

> the UINT64 is missing because there is no way to store it in some of the databases, it is always is casted to IN64, this may have changed in the meantime

One option, if it is really needed, is to store the value using two [PositiveBigIntegerField](https://docs.djangoproject.com/en/5.0/ref/models/fields/#positivebigintegerfield). What do you think ?
It could be possible to do when I will push the commits allowing to specify a custom data source (allowing to use a specific django model with specific read/write functions)

clavay, 2024-02-13:
In order to have a unique key, what do you think to change `value_class_choices` by 
```
    value_class_choices = (
        ("FLOAT32", "REAL, SINGLE or FLOAT32"),
        ("UNIXTIMEF32", "UNIXTIMEF32"),
        ("FLOAT64", "LREAL, FLOAT, DOUBLE or FLOAT64"),
        ("UNIXTIMEF64", "UNIXTIMEF64"),
        ("FLOAT48", "FLOAT48"),
        ("INT64", "INT64"),
        ("UINT64", "UINT64"),
        ("UNIXTIMEI64", "UNIXTIMEI64"),
        ("INT48", "INT48"),
        ("UNIXTIMEI32", "UNIXTIMEI32"),
        ("INT32", "INT32"),
        ("UINT32", "DWORD or UINT32"),
        ("INT16", "INT or INT16"),
        ("UINT16", "WORD UINT or UINT16"),
        ("INT8", "INT8"),
        ("UINT8", "UINT8"),
        ("BOOLEAN", "BOOL or BOOLEAN"),
    )
```

trombastic, 2024-02-16:
this is a good idea

PR #133 (2024-02-22): unique value class key for variable
close #103
closed 2024-02-22 by trombastic