pyscada/PyScada #117: API

Kimyasallar, 2024-01-30:
Hi all, that's me again with a new topic now :)

Is there a way through to get the variable data from an API rather than a device, if yes what is the practical one to achieve it?

Thank you.

Kimyasallar, 2024-01-30:
If no, what will be the best way to implement it IYO?

clavay, 2024-01-30:
Do you mean a web api? 
Using get or post request?

Kimyasallar, 2024-01-31:
Exactly it is.

ThingSpeak is my target. There is an endpoint like "https://api.thingspeak.com/channels/266256/fields/2.json" which can be assigned to a variable on pyscada?

clavay, 2024-02-02:
The plugin [PyScada-WebService](https://github.com/clavay/PyScada-WebService) can do it.

I will rewrite it because I did it a long time ago it is difficult to use. (remove the `WebServiceAction` to put it in the `WebServiceDevice` class)

But you can achieve what you want if you try this plugin.

Your url gives me : 
```
status	"404"
error	"Not Found"
```
So I cannot help you.

clavay, 2024-02-02:
I will try to rewrite this plugin next week.

Kimyasallar, 2024-02-05:
> I will try to rewrite this plugin next week.

I would appreciate a quick fix @clavay 🙏🏼🙏🏼🙏🏼

> The plugin [PyScada-WebService](https://github.com/clavay/PyScada-WebService) can do it.
>
> I will rewrite it because I did it a long time ago it is difficult to use. (remove the `WebServiceAction` to [...]

The url is an example that is given on the project ThingSpeak's page. The project is a common one used by lots of users.
If needed, there are two versions of the requests that are available.

Feed based:
https://api.thingspeak.com/channels/2409726/feeds.json

Field based: (variable based)
https://api.thingspeak.com/channels/2409726/fields/1.json

Rather than all, I am still having issues installing the current plugin to my system my friend.
I have tried too many ways to succeed
The link address "https://github.com/pyscada/PyScada-WebService/tarball/master" returns 404.
I don't have any instrucitons for git clone installation. 

And one last thing: On one of my efforts to succeed, I got the below error while migration:
AttributeError: module 'pyscada.hmi.models' has no attribute 'validate_html'

Can you shed some light on the path I should follow?

Thank you.

closed 2024-02-09 by clavay

Kimyasallar, 2024-02-09:
A new question:

Is there a way to make a donation to the project for this kind of effort to be sustainable?

clavay, 2024-02-09:
@trombastic : what do you think ?

Kimyasallar, 2024-02-09:
While both you thinking :) can I ask what is the "Web Service Action" class doing?

You have added with the update isn't it?

I have seen your reply but didn't have time to try what you said above. Now I upgraded Pyscada-Webservice and did what you said, unfortunately no value could be taken. :(

When I check the debug, it stays still at the below point:

[image]

clavay, 2024-02-10:
The update I made this week remove the Web Service Action class.

Please install the las version

Kimyasallar, 2024-02-10:
I really don't understand why I can't manage to do it at first glance while you are so fast to get results on the coding side.

This is a fresh OS install and modbus & webservice is up-to-date from git install but when I check plugins:

pip3 list | grep cada
PyScada                       0.8.2
pyscada-modbus         0.8.0
pyscada-webservice     0.8.0

and the error I got is:

TypeError: int() argument must be a string, a bytes-like object or a real number, not 'dict'

I have tried 6-7 different variable Value_class options but they all the same.

However, I had absolute faith that my attempt would be successful this time. :(

clavay, 2024-02-11:
Can you send me more logs?

Kimyasallar, 2024-02-11:
Sure. Here it is.

[image]

As I looked deeper into the system, it seems as on one of my trials of Value_class options, there was a read of "None" value yesterday. But as the install script of venv option installs only mariadb not postgres I didn't check what is written to db from webservice. Now I will look at the db from the terminal of any values written while the logs are shared with you herein.

Will share them as well asap. Hope you will find what my mistake is with ease. Thank you.

P.S.:
> @trombastic : what do you think ?
Any solutions? 👍

Kimyasallar, 2024-02-11:
Here is the db:

[image]

Kimyasallar, 2024-02-11:
Now something different has happened. I have changed the Value_class of the existing variable to bool again as there were some data on that class and it gave me the below error:

[image]

Kimyasallar, 2024-02-11:
But of course as the bgp fails more than 3 times the service stops.

[image]

clavay, 2024-02-11:
Your webservice variable configuration seems to be wrong as it select a dict to read, which is not possible for PyScada to store.

can you send me the device and variable config ?

But the unhandled exception is not normal, please try this branch to see if the exception is fixed : https://github.com/clavay/PyScada/tree/fixUpdateValuesChecks

clavay, 2024-02-11:
> > But the unhandled exception is not normal, please try this branch to see if the exception is fixed : https://github.com/clavay/PyScada/tree/fixUpdateValuesChecks
> 
> Do you want me to go on with a clean OS install?

No just install this branch and send me the logs.

Kimyasallar, 2024-02-12:
Here are the configs:

[image]

[image]

Now will go on with your branch. 🙏🏼

clavay, 2024-02-12:
The way I implement the json reading is : 
`field1 field2 [5] field3`

The fields needs to be space separated.
If there is a list, the index should be space separated also and the syntax is `[index]`

So in you case it should be `feeds [0] field1`

Kimyasallar, 2024-02-12:
The same :(

TypeError: int() argument must be a string, a bytes-like object or a real number, not 'dict'

Kimyasallar, 2024-02-12:
TOTALLY my mistake again! :((

Seems as it is working now. I am so sorry to bother for the latest part. I have now some read values. Let me check all in details.

Thank you very much.

clavay, 2024-02-12:
Good to know.

For your information I am thinking to implement a new way to store string for variables (without using dictionaries).

Kimyasallar, 2024-02-12:
I was going to ask what the dictionary class is doing there which we didn't use for webservice. (again without checking the implementation, just saw dictionary and dictionary item)

clavay, 2024-02-12:
Currently if you try to store a string to a variable, it is not possible to use the `RecordedData` table which can only store boolean, integer or float values.

I have added the `Dictionary` model to create (key,value) items.

So currently, if a device want to store a string for a variable : 
- if the variable hasn't dictionary : a dictionary is created,
- if the string does not exist in the dictionary, a new (key, value) is added, else the key of the string is used,
- the key is saved in the RecordedData table.

I am thinking about adding a [CharField](https://docs.djangoproject.com/en/5.0/ref/models/fields/#charfield) column to a new `RecordedDataNew` table to store strings.
This new `RecordedDataNew` table may have only one other column to store boolean, integer and float : a [FloatField](https://docs.djangoproject.com/en/5.0/ref/models/fields/#floatfield).
I am doing some test before implementing this.

Kimyasallar, 2024-02-12:
Will converting the json string values into integer and using as it is will be an easier solution by your side?

When I think of "What kind of situations will occur for webservices to retrieve data?", the only thing that comes to my mind is again values of integers or actually numbers rather than strings. Isn't it?

Kimyasallar, 2024-02-12:
Or will it be compute-expensive?

clavay, 2024-02-12:
What do you mean with 

> converting the json string values into integer

We can have webservice returning a string, for example `channel name` returns the string `m66v2` for https://api.thingspeak.com/channels/2409726/feeds.json

Kimyasallar, 2024-02-12:
Yes, we are retrieving strings from webservice json responses so rather than storing them as strings in DB maybe we can convert the digital values into integer or float and store as it is. Because any need for retrieving data from outside (json from webservice) shall include a measurement number in most cases, isn't it?

clavay, 2024-02-12:
Some protocols (OPC-UA, Bacnet) can return strings.

I don't understand how do you want to convert a string to a number without loosing the information of the text.
Or you create a dictionary and save the key, or you save the string directly.

Kimyasallar, 2024-02-12:
If protocols like OPC or Bacnet returns data as string and there is also the matter of losing sensibility of the data, then you are right. Forget about my question. :)

Kimyasallar, 2024-02-13:
Hi again @clavay 🖐🏼

> I updated the [PyScada-WebService](https://github.com/pyscada/PyScada-WebService) plugin.
> 
> Install it and configure a device with the URL set to https://api.thingspeak.com/channels/2409726/feeds.json
> 
> for example, create variables with the path set to :
> 
> * channel name
> * feeds [0] created_at
> 
> Look at the result in the variable dictionary.
> 
> Create a `control panel `with a `control item` for each `variable` (`control item` type set as `display value`) Add it to a `page` using a `widget`, create a `view` with this `page` and look at the result in the user interface.
> 
> Reopen the issue if you see a bug or if you have a question.

with this path implementation, how is it possible to retrive feed at all? I mean, by selecting "feeds[0].field1" as the variable, I am not able to read the consequtive measurements of a field or the answer will be "I am doing something wrong again" ?

clavay, 2024-02-13:
I had the same issue with the infoclimat API.
I solved it with the handler [infoclimat](https://github.com/clavay/PyScada-WebService/blob/main/pyscada/webservice/devices/infoclimat.py#L92-L186)

Look at this handler, try do do your handler and let me know if you need help.

Here is the output of the API to understand and let you adapt it to your case:
```
{
    "status": "OK",
    "errors": [],
    "data": [],
    "stations": [
        {
            "id": "000PS",
            "name": "T\u00e9thieu",
            "latitude": 43.75,
            "longitude": -0.983,
            "elevation": 20,
            "type": "static",
            "license": {
                "license": "CC BY",
                "url": "https:\/\/creativecommons.org\/licenses\/by\/2.0\/fr\/",
                "source": "infoclimat.fr",
                "metadonnees": "https:\/\/www.infoclimat.fr\/stations\/metadonnees.php?id=000PS"
            }
        }
    ],
    "metadata": {
        "temperature": "temperature,degC",
        "pression": "mean sea level pressure,hPa",
        "humidite": "relative humidity,%",
        "point_de_rosee": "dewpoint,degC",
        "visibilite": "horizontal visibility,m",
        "vent_moyen": "mean wind speed,km\/h",
        "vent_rafales": "wind gust,km\/h",
        "vent_direction": "wind direction,deg",
        "pluie_3h": "precipitation over 3h,mm",
        "pluie_1h": "precipitation over 1h,mm",
        "neige_au_sol": "snow depth,cm",
        "nebulosite": "Ncloud cover,octats",
        "temps_omm": "present weather,http:\/\/www.infoclimat.fr\/stations-meteo\/ww.php"
    },
    "hourly": {
        "000PS": [
            {
                "id_station": "000PS",
                "dh_utc": "2024-01-01 00:00:00",
                "temperature": "8",
                "pression": "1018.8",
                "humidite": "97",
                "point_de_rosee": "7.5",
                "vent_moyen": "1.5",
                "vent_rafales": null,
                "vent_direction": "181",
                "pluie_3h": "0",
                "pluie_1h": "0"
            },
            {
                "id_station": "000PS",
                "dh_utc": "2024-01-01 00:10:00",
                "temperature": "8.1",
                "pression": "1018.6",
                "humidite": "97",
                "point_de_rosee": "7.6",
                "vent_moyen": "1.5",
                "vent_rafales": null,
                "vent_direction": "173",
                "pluie_3h": null,
                "pluie_1h": null
            },
[... 1797 lines cut ...]
                "temperature": "6.8",
                "pression": "1017.1",
                "humidite": "95",
                "point_de_rosee": "6.1",
                "vent_moyen": "0",
                "vent_rafales": null,
                "vent_direction": null,
                "pluie_3h": null,
                "pluie_1h": null
            }
        ],
        "_params": [
            "temperature",
            "pression",
            "humidite",
            "point_de_rosee",
            "vent_moyen",
            "vent_rafales",
            "vent_direction",
            "pluie_3h",
            "pluie_1h"
        ]
    }
}
```

clavay, 2024-02-13:
Please open an issue in the corresponding plugin if needed (`WebService` here)

Kimyasallar, 2024-02-13:
But the webservice plugin repo doesn't have issues option there.

For the implementation part, I have seen enedis.py before and tried. Now I have tried with infoclimat.py. But not able to sort the code depending on the TS API. Will go on further tomorrow. Thank you @clavay

Kimyasallar, 2024-02-14:
@clavay,

How to achive with TS, can you help me some?

Kimyasallar, 2024-02-14:
I have seen "Issue" on PyScada/WebService repo in the morning but now I was going to open a new issue with this TS topic there. But I can't see it now.

clavay, 2024-02-14:
you need to open the issue on my repo : https://github.com/clavay/PyScada-WebService