pyscada/PyScada #144: Testing of OPC UA and other communication

nadim4114, 2024-05-05:
Hi Admins,
I have worked in Industrial Automation field and SCADA application development for 10+ years and recently started working on Python and Django framework.

With my experience with SCADA and also Django fw, I am interested in contributing to opensource Scada PyScada and as part of that journey, I want to test the PyScada communication with various devices using various communication protocols and provide feedback for improvement.

I have experience on OPC UA and Modbus so I can help testing. I have various devices available with me to test communication.

Pls let me know how can I start?

clavay, 2024-05-07:
Hello,

Good to know that you are interested to test devices.
Are you interested into improve the code ?

First you can install PyScada and the plugin you need (PyScada-Modbus, PyScada-OPCUA...) and try to communicate (read and write) with your devices.

The documentation available is [here](pyscada.readthedocs.io)
First you may want to improve the documentation to install/configure/use the software.
There are many things missing in the documentation, especially in the configuration (admin/backend) and the use (user/frontend).
And then let us know what trouble you have.

Then if you want quick changes, the best is to write pull requests, otherwise it will depend on the time and the devices we have to resolve yours issues.

nadim4114, 2024-05-09:
I see pyScada very different from other Scada. Most Scada are WYSIWYG designer but I see the WYSIWYG functionality missing. I want to work on a WYSIWIG designer which makes Scada very user friendly. What is your opinion on this WYSIWYS designer?

Definitely the documentation needs to be improved. I can update the docs as I setup the Scada. I have basic experience of sphinx for documentation.

clavay, 2024-05-13:
For the documentation, a good idea could be a tutorial showing in 30 minutes how to :

- install pyscada and plugins,
- configure 2 instruments (using 2 different protocols) with input or output variables,
- create a simple user interface (with chart, text, control panel),
- create some events.

For the instrument communication, the best could be to start with the modbus protocol, which is better integrated.
Can you start installing the plugin and try to communicate (input/output) and tell me what you want to improve.
Next you can try with OPC UA instruments.

For the WYSIWYG implementation, do you mean to be able to edit the user interface directly in the user interface with an _edition mode_ ?
One option I was thinking about, is to implement a way to build the user interface using [django CMS](https://www.django-cms.org/en/) or [wagtail](https://wagtail.org).
I don't know which one would be the best between these CMS or an internal development.

trombastic, 2024-05-13:
Sounds like a good plan to me, please open a new issue for real bugs, small questions should be asked here in my opinion.
Regards,
Martin

clavay, 2024-05-13:
Sounds good for me.
Improvements should also be in a separated issue, i think.

nadim4114, 2024-05-14:
Presently I am setting up the pyscada on my VM. 
I will create a separate issue to ask for help to setup. 
Once PyScada is setup, I will update the document in detail about setup procedure.

clavay, 2024-08-28:
I think we can close this issue, unless you have other questions.

clavay, 2024-10-17:
Please reopen if needed.