pyscada/PyScada #145: Installing and Getting Started PyScada

nadim4114, 2024-05-14:
Hey Guys, I am installing PyScada in Ubuntu 20.04 LTS version.
I have created this issue for any help and support required for setup if I face any problem.

I am following documentation and found the below two instructions to setup PyScada : 

1. https://pyscada.readthedocs.io/en/main/installation.html

I am stuck at this place : https://pyscada.readthedocs.io/en/main/installation.html#initialize-database-and-copy-static-files

I am getting below errors : 
root@ubuntu:/var/www/pyscada/PyScadaServer# sudo -u pyscada python3 manage.py collectstatic

You have requested to collect static files at the destination
location as specified in your settings.

This will overwrite existing files!
Are you sure you want to do this?

Type 'yes' to continue, or 'no' to cancel: yes
Traceback (most recent call last):
  File "manage.py", line 22, in <module>
    main()
  File "manage.py", line 18, in main
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.8/dist-packages/django/core/management/__init__.py", line 442, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.8/dist-packages/django/core/management/__init__.py", line 436, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/usr/local/lib/python3.8/dist-packages/django/core/management/base.py", line 412, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/usr/local/lib/python3.8/dist-packages/django/core/management/base.py", line 458, in execute
    output = self.handle(*args, **options)
  File "/usr/local/lib/python3.8/dist-packages/django/contrib/staticfiles/management/commands/collectstatic.py", line 209, in handle
    collected = self.collect()
  File "/usr/local/lib/python3.8/dist-packages/django/contrib/staticfiles/management/commands/collectstatic.py", line 135, in collect
    handler(path, prefixed_path, storage)
  File "/usr/local/lib/python3.8/dist-packages/django/contrib/staticfiles/management/commands/collectstatic.py", line 368, in copy_file
    if not self.delete_file(path, prefixed_path, source_storage):
  File "/usr/local/lib/python3.8/dist-packages/django/contrib/staticfiles/management/commands/collectstatic.py", line 278, in delete_file
    if self.storage.exists(prefixed_path):
  File "/usr/local/lib/python3.8/dist-packages/django/core/files/storage/filesystem.py", line 165, in exists
    return os.path.lexists(self.path(name))
  File "/usr/local/lib/python3.8/dist-packages/django/contrib/staticfiles/storage.py", line 39, in path
    raise ImproperlyConfigured(
django.core.exceptions.ImproperlyConfigured: You're using the staticfiles app without having set the STATIC_ROOT setting to a filesystem path.

There is also one more tutorial :
https://pyscada.readthedocs.io/en/main/quick_install.html

Can you tell me which one tutorial should I follow to start with and what is the difference between both of them?

nadim4114, 2024-05-14:
I tried below tutorial also and got errors.
https://pyscada.readthedocs.io/en/main/quick_install.html

Log file is attached here.
[logs_install.txt](https://github.com/pyscada/PyScada/files/15302637/logs_install.txt)

clavay, 2024-05-14:
The documentation from https://pyscada.readthedocs.io/en/main/installation.html is not up to date.
How did you get this url ?
It should be deleted or updated (to explain a manual installation).

For the log from the script install :

- You get this error :   `error: invalid command 'bdist_wheel'` 
As said [here](https://stackoverflow.com/questions/37029385/cant-build-wheel-error-invalid-command-bdist-wheel) can you try to install `wheel` in the virtual environment ?
```
source /home/pyscada/.venv/bin/activate
sudo -u pyscada -E env PATH=${PATH} pip install wheel
```

- You get this error : `django.db.utils.NotSupportedError: MariaDB 10.4 or later is required (found 10.3.39).`
As said [here](https://docs.djangoproject.com/en/5.0/releases/4.2/#dropped-support-for-mariadb-10-3) django 4.2 needs MariaDB>10.4.
Upgrade your ubuntu version or install mariadb manually to get the version > 10.4

clavay, 2024-08-28:
If the problem is solved, please close the issue.

clavay, 2024-10-17:
Please reopen if needed.

closed 2024-10-17 by clavay