pyscada/PyScada #154: Change default time range on views

Aliefe2002, 2024-07-01:
I have added a lot of variables and charts on a view so now it is very laggy. Default time range is 2 hours so a lot of data is received when page is loading. I want to reduce that 2 hours to 30 min or 10 min so that my page can load easier. I have found 3 locations that promise to be the default time range however they weren't. Could you help me with that?

[image]
this is hmi/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js. 
At first i thought that when page is loading, PlusTwoHours button is pressed and this button makes start of date range 2 hours ago, however this change didn't work

[image]
this is again on the same file

[image]
again same file and this looked most promisin however changes still didn't affect anything. 

So where can i change the default time range for when page first loads?
Thanks

clavay, 2024-07-02:
Hello,
It was an improvement planned.
It should be fixed in #155.

In this PR I added a view option to set the default time delta you want (default to 2 hours).
After updating to this new PyScada version (v0.8.3) you need to apply migrations and copy static files (see point 2 [here](https://pyscada.readthedocs.io/en/main/plugin_install.html#list-pyscada-plugin-installed))

closed 2024-07-02 by trombastic

Aliefe02, 2024-07-29:
Hi, i know you pushed an update for this, thx for that. However for people using previous versions, turns out first 2 screenshots i provided were the right place to change default time. However i was modifying the wrong file, i was modifying the file inside pyscada/hmi/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js.

turns out this wasnt the file served. The file i needed to modify was located at 
/var/www/pyscada/http/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js

For anyone who use previous version, you can change default time by changing those lines

Aliefe2002, 2024-07-29:
Hi, i know you pushed an update for this, thx for that. However for people using previous versions, turns out first 2 screenshots i provided were the right place to change default time. However i was modifying the wrong file, i was modifying the file inside pyscada/hmi/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js.

turns out this wasnt the file served. The file i needed to modify was located at
/var/www/pyscada/http/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js

For anyone who use previous version, you can change default time by changing those lines

clavay, 2024-08-28:
To answer you comment, the django `collectstatic` command copy the static files (JavaScript, CSS...) from static directories (as `/home/pyscada/.venv/....../pyscada/hmi/static/`) to `STATIC_ROOT` (defined in `settings.py` as  `/var/www/pyscada/http/static/`, see [here](https://docs.djangoproject.com/en/5.1/ref/contrib/staticfiles/#django-admin-collectstatic)).

So if you manually modify the `/home/pyscada/.venv/....../pyscada/hmi/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js` file, you can copy it to  `/var/www/pyscada/http/static/pyscada/js/pyscada/pyscada_v0-7-0rc14.js` using the `collectstatic` command (see point 2 [here](https://pyscada.readthedocs.io/en/main/plugin_install.html#list-pyscada-plugin-installed)).