pyscada/PyScada #156: This error originates from a subprocess, and is likely not a problem with pip.

e41c9fp9duckcom, 2024-07-03:
error: subprocess-exited-with-error

× python setup.py develop did not run successfully.
│ exit code: 1
╰─> [29 lines of output]
    running develop
    /home/sct/anaconda3/envs/pscada/lib/python3.11/site-packages/setuptools/command/develop.py:40: EasyInstallDeprecationWarning: easy_install command is deprecated.
    !!
    
            ********************************************************************************
            Please avoid running ``setup.py`` and ``easy_install``.
            Instead, use pypa/build, pypa/installer or other
            standards-based tools.
    
            See https://github.com/pypa/setuptools/issues/917 for details.
            ********************************************************************************
    
    !!
      easy_install.initialize_options(self)
    /home/sct/anaconda3/envs/pscada/lib/python3.11/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
    !!
    
            ********************************************************************************
            Please avoid running ``setup.py`` directly.
            Instead, use pypa/build, pypa/installer or other
            standards-based tools.
    
            See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
            ********************************************************************************
    
    !!
      self.initialize_options()
    running egg_info
    error: Cannot update time stamp of directory 'PyScada.egg-info'
    [end of output]

note: [image]

e41c9fp9duckcom, 2024-07-03:
do you have any setup doct?

clavay, 2024-07-03:
Maybe you are running the command as the pyscada user but the user don't have right to write to all files and sub-directories ?

Please provide more information about what you are doing.

clavay, 2024-08-28:
Any news ?
If not please close the issue.

clavay, 2024-10-17:
Please reopen if needed.

closed 2024-10-17 by clavay