pyscada/PyScada #162: OneWire Device is not reading any values

joliperj, 2024-09-04:
[04/Sep/2024 14:33:00] DEBUG [pyscada.utils.scheduler:537] process pyscada.onewire-1 started
[04/Sep/2024 14:33:01] ERROR [pyscada.models:1913] update_value is depreceated, use update_values instead. Variable 1 - Temp_Sens_Var1.
[04/Sep/2024 14:33:01] DEBUG [pyscada.models:2015] 1 - Temp_Sens_Var1 updated 1 - 0 values
[04/Sep/2024 14:33:01] ERROR [pyscada.utils.scheduler:923] pyscada.onewire-1, unhandled exception
Traceback (most recent call last):
  File "/home/pyscada/.venv/lib/python3.11/site-packages/pyscada/utils/scheduler.py", line 822, in run
    status, data = self.loop()
                   ^^^^^^^^^^^
  File "/home/pyscada/.venv/lib/python3.11/site-packages/pyscada/utils/scheduler.py", line 1522, in loop
    tmp_data = self.device.request_data()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pyscada/.venv/lib/python3.11/site-packages/pyscada/onewire/device.py", line 58, in request_data
    output.append(item.create_recorded_data_element())
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Variable' object has no attribute 'create_recorded_data_element'
[04/Sep/2024 14:33:01] DEBUG [pyscada.utils.scheduler:1046] Process pyscada.onewire-1(3843) is stopped
[04/Sep/2024 14:33:10] DEBUG [pyscada.utils.scheduler:549] process 219 is dead
[04/Sep/2024 14:33:27] ERROR [pyscada.utils.scheduler:1151] process pyscada.onewire-1 failed 3 times

You can see below that OneWire Temperature Sensor is working as expected,
[image]

clavay, 2024-09-04:
What are the PyScada version and PyScada-OneWire version you have installed ?

It seems you don't have the last PyScada-OneWire version.

joliperj, 2024-09-05:
PyScada version: 0.8.3
PyScada-OneWire version: 0.8.0

joliperj, 2024-09-05:
So what I figured out is that on path /home/pyscada/.venv/lib/python3.11/site-packages/pyscada/onewire the files are not in the latest version. 
The steps I used in order to install this plugin are the mentioned on this link https://pyscada.readthedocs.io/en/main/plugin_install.html

I replaced the files manually and it worked. Can it be that the installation method is wrong?

clavay, 2024-09-05:
what method do you choose to download the plugin ? Cloning the repository or downloading the zip ?

I tried both with onewire plugin and I get the last version.

When cloning the repository, if you did it a long time ago, you must fetch and pull the last changes.

Please close the issue if it is good for you.

joliperj, 2024-09-06:
I cloned the repository. I actual did that 2 weeks before. But anyway I managed to fix that manually.

joliperj, 2024-09-06:
Correct version for OneWire plugin repo had to be installed.

closed 2024-09-06 by joliperj