pyscada/PyScada #173: Documentation regarding uninstalling PyScada from Ubuntu 18.04

joeldevasia, 2024-10-29:
I recently tried installing PyScada in my Ubuntu 18.04 Desktop. But I couldn't complete the installation in Venv as well as Docker. In Docker the installation script stuck after starting the MySQL Container and when checking the status of the container using `docker ps`. I could see that the MySQL Container was restarting indefinitely. So I removed all Containers and images and tried the Venv method, during installation there were errors regarding missing imports, so after some googling I thought it might be due to older python version or something like that. I tried installing the missing imports but still the issue didn't go. So I decided to completely remove PyScada. To my surprise there isn't anything about un-installation in the docs. I want to remove it particularly because after installation Ubuntu takes soo much time to boot, probably because of incomplete installation. I tried disabling the PyScada service but still it takes time to boot. Maybe it is able to boot because it's timing out, IDK.

I'd appreciate it if anyone could help me with the removal process. After that I'll try installing it in another desktop with newer Ubuntu.

joeldevasia, 2024-10-30:
The desktop taking time to boot was due to other issues and not related to PyScada. But still it would be great if there were documentation regarding uninstall procedure. 
The boot time issue was due to failed installation of PyScada through docker. The Docker messed with the Network configurations due to which no networks were inaccessible through the desktop, be it Wifi, Ethernet, USB Tethering. I tried fixing but missed something in the process.
Sorry for the trouble.

clavay, 2024-10-31:
If you give me more details about the docker or venv installation, maybe I can help you.

Ok, I will write a uninstallation documentation.

clavay, 2024-11-19:
Can you try this to uninstall : https://github.com/clavay/PyScada/blob/uninstallDoc/docs/uninstall.rst

joeldevasia, 2024-11-19:
Okay thanks

clavay, 2024-12-09:
Reopen the issue if needed.

closed 2024-12-09 by clavay