pyscada/PyScada #189: PyScada Docker Installation Issue

joeldevasia, 2024-12-14:
I am trying to install PyScada in my Ubuntu 20.04 machine through the docker method. After starting the setup process, it seems to stuck at this output `Creating mysql ... done`. Running `docker ps -a` shows that the **mysql:latest**'s container is restarting indefinitely. When checking the logs of that container, I get this.

2024-12-14 12:35:58+00:00 [Note] [Entrypoint]: Entrypoint script for MySQL Server 9.1.0-1.el9 started.
2024-12-14 12:35:59+00:00 [Note] [Entrypoint]: Switching to dedicated user 'mysql'
2024-12-14 12:35:59+00:00 [Note] [Entrypoint]: Entrypoint script for MySQL Server 9.1.0-1.el9 started.
'/var/lib/mysql/mysql.sock' -> '/var/run/mysqld/mysqld.sock'
2024-12-14T12:35:59.979287Z 0 [System] [MY-015015] [Server] MySQL Server - start.
2024-12-14T12:36:00.359909Z 0 [System] [MY-010116] [Server] /usr/sbin/mysqld (mysqld 9.1.0) starting as process 1
2024-12-14T12:36:00.370561Z 1 [System] [MY-013576] [InnoDB] InnoDB initialization has started.
2024-12-14T12:36:03.088398Z 1 [System] [MY-013577] [InnoDB] InnoDB initialization has ended.
mysqld: Table 'mysql.plugin' doesn't exist
2024-12-14T12:36:03.273012Z 0 [ERROR] [MY-010735] [Server] Could not open the mysql.plugin table. Please perform the MySQL upgrade procedure.
2024-12-14T12:36:03.273407Z 0 [Warning] [MY-010441] [Server] Failed to open optimizer cost constant tables
2024-12-14T12:36:03.273651Z 0 [Warning] [MY-010441] [Server] Failed to open optimizer cost constant tables
2024-12-14T12:36:03.273842Z 0 [Warning] [MY-010441] [Server] Failed to open optimizer cost constant tables
2024-12-14T12:36:03.274144Z 0 [Warning] [MY-010441] [Server] Failed to open optimizer cost constant tables
2024-12-14T12:36:03.274346Z 0 [Warning] [MY-010441] [Server] Failed to open optimizer cost constant tables
2024-12-14T12:36:03.274517Z 0 [Warning] [MY-010441] [Server] Failed to open optimizer cost constant tables
2024-12-14T12:36:03.274674Z 0 [Warning] [MY-010441] [Server] Failed to open optimizer cost constant tables
2024-12-14T12:36:03.627891Z 0 [Warning] [MY-010015] [Repl] Gtid table is not ready to be used. Table 'mysql.gtid_executed' cannot be opened.
2024-12-14T12:36:03.719101Z 0 [Warning] [MY-010015] [Repl] Gtid table is not ready to be used. Table 'mysql.gtid_executed' cannot be opened.
2024-12-14T12:36:03.783904Z 0 [Warning] [MY-010068] [Server] CA certificate ca.pem is self signed.
2024-12-14T12:36:03.784168Z 0 [System] [MY-013602] [Server] Channel mysql_main configured to support TLS. Encrypted connections are now supported for this channel.
2024-12-14T12:36:03.806938Z 0 [Warning] [MY-011810] [Server] Insecure configuration for --pid-file: Location '/var/run/mysqld' in the path is accessible to all OS users. Consider choosing a different directory.
2024-12-14T12:36:03.807656Z 0 [Warning] [MY-010441] [Server] Failed to open optimizer cost constant tables
2024-12-14T12:36:03.807942Z 0 [ERROR] [MY-013129] [Server] A message intended for a client cannot be sent there as no client-session is attached. Therefore, we're sending the information to the error-log instead: MY-001146 - Table 'mysql.component' doesn't exist
2024-12-14T12:36:03.808098Z 0 [Warning] [MY-013129] [Server] A message intended for a client cannot be sent there as no client-session is attached. Therefore, we're sending the information to the error-log instead: MY-003543 - The mysql.component table is missing or has an incorrect definition.
2024-12-14T12:36:03.808324Z 0 [ERROR] [MY-000067] [Server] unknown variable 'default-authentication-plugin=mysql_native_password'.
2024-12-14T12:36:03.809216Z 0 [ERROR] [MY-010119] [Server] Aborting
2024-12-14T12:36:05.274991Z 0 [System] [MY-010910] [Server] /usr/sbin/mysqld: Shutdown complete (mysqld 9.1.0)  MySQL Community Server - GPL.
2024-12-14T12:36:05.275089Z 0 [System] [MY-015016] [Server] MySQL Server - end.

Seems some tables are not there. I would highly appreciate if someone would help me solve this issue. 
Thankyou in advance.

felando1984, 2024-12-15:
I encountered the same issue yesterday. Thank you, **joeldevasia**, for your post!

I used the "docker-compose.yml" from [docker-compose.yml-tmp](https://github.com/pyscada/PyScada/blob/main/docker/docker-compose.yml-tmp)".

[docker-compose.zip](https://github.com/user-attachments/files/18140289/docker-compose.zip)

I tested all three template projects, and the results were the same.  [templ](https://github.com/pyscada/PyScada/tree/main/tests)
Additionally, I tested the template project with only the PyScada image from this repository. (https://hub.docker.com/r/zchfly/docker_pyscada)

I also tried various versions of the MySQL image, but the issue persisted across all of them.
Here’s a screenshot of the error for reference:
[image]

I’m looking forward to any assistance with resolving this issue.

felando1984, 2024-12-16:
> Are you using the `install.sh` script to install PyScada on a linux machine ? (choosing the docker method at the first script question)

If run install.sh, it seems more errors. If you could use docker for PyScada, could you help update Dockerfile so we could just build docker file by ourselves or share the docker image in dockerhub?
[image]

joeldevasia, 2024-12-17:
> Are you using the `install.sh` script to install PyScada on a linux machine ? (choosing the docker method at the first script question)

Yes, after running the command I chose docker.

ironmouh, 2024-12-18:
The issue seems to come from incompatible versions between MySQL and Django as well as an issue of MySQL latest image:

- When pulling latest MySQL image, the container stays in starting mode
- When updating the docker-compose.yml-tmp and Dockerfile-tmp to force the recommended version commented in Dockerfile saying to pull MySQL 5.7, there is an issue with Django version 4.2 that does not support MySQL 5.7 but only MySQL 8 and above

@clavay, If you have a working version, could you please juste provide us the versions of MySQL and Django that you are using ?

clavay, 2024-12-18:
Can you try [this branch from my fork](https://github.com/clavay/PyScada/tree/dockerInstall) ?

It worked for me with the changes from the last pushed commit.

I'm not a docker specialist but I think it is better to clean your docker (builder, container, image, volume) before installing using the `install.sh` script.

felando1984, 2024-12-18:
I tried the new branch, but I didn't notice any differences. Therefore, I decided to stop using Docker. Instead, I tried installing PyScada on a clean Ubuntu 20.04.4 VM image. The result is attached, and it seems that the installation of PyScada using install.sh failed. Should I install MySQL and Django manually before running install.sh? Could you help me with a quick check? Thank you.
[InstallError.docx](https://github.com/user-attachments/files/18189307/InstallError.docx)

felando1984, 2024-12-19:
God bless! I finally managed to successfully use PySCADA in my VM image. At least it works for Modbus TCP, as shown in the attachment. I am planning to use OPC UA, but the plugin doesn’t seem to work. It doesn’t even connect to my server, despite being configured. I will conduct more detailed tests and seek help if I still can’t resolve the issue. For now, the setup issue is manageable. Thank you very much for your help and reminder.

[
[image]
](url)

clavay, 2024-12-20:
Good news.
Did you install PyScada using docker or in a venv on your VM ?

For OPC-UA if you have an issue, try connect to your device using the [asyncua](https://github.com/FreeOpcUa/opcua-asyncio) library (use by the PyScada-OPCUA plugin) and show me your script.
Please create an issue in the [PyScada-OPCUA](https://github.com/pyscada/PyScada-OPCUA) repository if needed.

felando1984, 2024-12-20:
At last I chose to install pyscada in a venv on my VM and it works. Thank you for your suggestion about asyncua, I created my issue there.

joeldevasia, 2024-12-22:
Finally, I was able to install PyScada using Docker method. I set Django version to 4.1 in `setup.py` and replaced `mysql:latest` with `mysql:5.7` in Dockerfile-tmp and docker-compose.yml-tmp. But still I was getting this error.
`ModuleNotFoundError: No module named 'pkg_resources'`
After some googling, I added this command to pyscada dockerfile
`RUN apt install python-setuptools`

Thankyou everyone for your support. Now I just need to figure out how to use PyScada :)

closed 2024-12-22 by joeldevasia